heitzmann/gdspy #108: get points out of a polygonset object

Johny-Leo, 2019-11-23:
Hi,

I am a newbie to gdspy and trying to use it along with gdsCAD. I would like to know how to convert Boundary object in gdsCAD to polygonset object in gdspy and vice versa.

The reason being that gdsCAD does not support boolean operations so I have to call gdspy to do it and then import the points from polygonset in gdspy to boundary in gdsCAD.

Thank you!

heitzmann, 2019-11-25:
@Johny-Leo, I'm not entirely sure because I've never used gdsCAD, but it seems you should be able to use `gdspy.Polygon(gdscad_boundary.points)`.

Just out of curiosity, what are the features from gdsCAD that you miss in Gdspy?

Johny-Leo, 2019-11-25:
Hi, Lucas, thank you very much for the reply. I also figured it out. To answer your question about gdsCAD, it basically is a module built on gdspy, I am not completely experienced with gdspy to make a comparison, my feeling of gdsCAD is that the control of Elements (collection of boundaries and paths), Cell, and  Layout are quite straightforward, although it has been out of update for years already.

Johny-Leo, 2019-11-25:
For anyone who also is wondering about this question, please check my file boolean_test.py on https://github.com/Johny-Leo/GDS-playground.git.

closed 2019-11-25 by Johny-Leo