heitzmann/gdspy #110: Layerwise boolean

villadsegede, 2019-11-27:
Hello,

Thanks for an excellent library!

Just one issue I have encountered: when doing boolean operations, it seems not to take the layer property into account. As an example, I have a Cell with polygons in two layers in it, and I want to 'and' it with a fillet rectangle. The output is the 'and' operation performed on all layers and then merged into one polygon in the same layer. Is there a way to preserve the layer information of the original cell?

Wouldn't it make more sense to 
1) preserve the layers of one of the input cell/polygons and use the other for the boolean operation?
2) Perform the boolean on a layer-by-layer basis? (or at least have it as an option)

Cheers,
Villads

heitzmann, 2019-11-27:
@villadsegede, Thank you for the feedback!

It's very hard to predict what the user wants to do (preserving or not layers or datatype or both, and in which operand). In such cases I try to opt for the most general or simple solution, so that specific behavior can be implemented on a higher level.

In the this case, to have a layer-wise operation, you can pass the polygons directly to `boolean`, instead of the whole cell:

```python
cell_polys = my_cell.get_polygons(by_spec=True)
bool_layer_polys = [poly for key, poly in cell_polys.items() if key[0] == bool_layer]
result = gdspy.boolean(bool_layer_poly, other_polys, operation)
```

(This can be greatly simplified if there is only 1 datatype in the desired layer.)

Of course it's possible do something similar to `other_polys` and to iterate over the layers/datatypes to get whatever combination you need.

villadsegede, 2019-11-28:
Fair point! Intuitively I expected the operation to be per datatype, but with the `get_polygons` method, it does make sense to keep the boolean as it is :)

closed 2019-11-28 by villadsegede