heitzmann/gdspy #111: ValueError: Operands could not be broadcast together with shapes

dwaardenburg, 2019-11-29:
Dear heitzmann,

I have made a piece of code that gives me quite a weird error.
Would you have any clue where this error could come from?

This is the code where it is going wrong:
    path_list.append(gd.FlexPath(to_point, to_end_width))
    j = 0
    for i in range(1, path_length - 1):
        if new_path_point_list:
            if path[i].tolist() == new_path_point_list[0].tolist():
                path_list[j].segment(path[i], width = path_widths[j])
                path_list.append(gd.FlexPath(path[i], path_widths[j]))
                np.delete(new_path_point_list, [0])
                j += 1
            else:
                path_list[j].segment(path[i], width = path_widths[j])
        else:
            path_list[j].segment(path[i], width = path_widths[j])
    path_list[-1].segment(from_point, width = from_end_width)

And this is the error:
File "/anaconda3/lib/python3.7/site-packages/gdspy/__init__.py", line 6648, in write_cell
    self._outfile.write(cell.to_gds(self._res, timestamp))
  File "/anaconda3/lib/python3.7/site-packages/gdspy/__init__.py", line 4878, in to_gds
    data.extend(path.to_gds(multiplier) for path in self.paths)
  File "/anaconda3/lib/python3.7/site-packages/gdspy/__init__.py", line 4878, in <genexpr>
    data.extend(path.to_gds(multiplier) for path in self.paths)
  File "/anaconda3/lib/python3.7/site-packages/gdspy/__init__.py", line 3103, in to_gds
    return self.to_polygonset().to_gds(multiplier)
  File "/anaconda3/lib/python3.7/site-packages/gdspy/__init__.py", line 3070, in to_polygonset
    polygons = self.get_polygons(True)
  File "/anaconda3/lib/python3.7/site-packages/gdspy/__init__.py", line 2866, in get_polygons
    pa = pts[:-1] + vn * (sign * 0.5 * self.widths[:-1, kk:kk + 1])
ValueError: operands could not be broadcast together with shapes (4,2) (5,1)

dwaardenburg, 2019-11-29:
The path[i] variables are points. The code runs fine if I only use the first and last line of this segment.

heitzmann, 2019-12-02:
@dwaardenburg I can look into it to try to figure out what's wrong, but I'm quite busy this week, so it could be a few days. In the mean time, do you have by any chance a complete example so that I can reproduce the bug myself?

dwaardenburg, 2019-12-06:
So far it seems that the flexpath.segment method might just continue if the point is a list of points. I thought that was an option. Let me check tomorrow if that was the case.

heitzmann, 2019-12-09:
[`FlexPath.segment`](https://gdspy.readthedocs.io/en/stable/reference.html#gdspy.FlexPath.segment) accepts only a single (x, y) point, not a list... is that the issue? From your code it seemed you were already passing single points, but you'd have to make sure path[i] and path_widths[i] are what you expect.

Let me know if you find out anything else...

heitzmann, 2019-12-19:
@dwaardenburg, I believe I found the problem (completely by chance while working on something else): the line `path_list.append(gd.FlexPath(to_point, to_end_width))` initializes `FlexPath` with `to_point`, which I assume is a size-2 vector. However, the first argument of `FlexPath` must be an [N×2 array](https://gdspy.readthedocs.io/en/stable/reference.html#flexpath).

I believe if you replace that line with `path_list.append(gd.FlexPath([to_point], to_end_width))` the issue will be solved. Could you give it a try?

This is a common enough use case that I should probably include as a special case for the first argument.

heitzmann, 2020-08-02:
It's been a while, but I believe this commit solves the issue.

closed 2021-01-22 by heitzmann