heitzmann/gdspy #113: What is the recommended alternative to GdsLibrary.extract

ncFoCo, 2019-12-20:
I just saw the note regarding the GdsLibrary.extract method being deprecated eventually. Is the cells dictionary the recommended accessor for a cell object?

heitzmann, 2019-12-20:
@ncFoCo `GdsLibrary.extract` was used only to add a cell from the library to `gdspy.current_library`, but that is also deprecated now. The better way to manage libraries is to simply use `GdsLibrary.add`, which, as a bonus, can also handle dependencies.

Instead of:
```python
# Load a GDSII file
imported_lib = gdspy.GdsLibrary(infile="mylib.gds")

# Extract a cell and implicitly add it to gdspy.current_library (without dependencies)
imported_lib.extract("SOME_CELL")

# Implicitly save gdspy.current_library
gdspy.write_gds("newlib.gds")
```
this is more explicitly and less error-prone:
```python
# Load a GDSII file
imported_lib = gdspy.GdsLibrary(infile="mylib.gds")

# Create a new library and add imported cell (including dependencies)
new_lib = gdspy.GdsLibrary()
new_lib.add(imported_lib.cells["SOME_CELL"])

# Save new library
new_lib.write_gds("newlib.gds")
```

ncFoCo, 2019-12-20:
@heitzmann That answers my question. Thanks for the quick response.

closed 2019-12-20 by ncFoCo