heitzmann/gdspy #114: `gdspy.current_library` might not be useless when we use an `GdsLibrary` instance.

mshrn, 2019-12-25:
Hello @heitzmann , I appreciate this useful interface of gds files. However, I got in a trouble when I use an `GdsLibrary` instance. I read your code of `library.py` and may understand how to solve the problem. However I want to know what your intention is, so I open an issue. 
I wrote a code on jupyter notebook as follows:
```
import gdspy

lib = gdspy.GdsLibrary()

cell = gdspy.Cell("main")
lib.add(cell)
rect = gdspy.Rectangle((0, 0), (2, 1), layer=3)
rect2 = gdspy.Rectangle((0.3, 0.3), (1.7, 0.7), layer=3)
cell.add(rect)
cell.add(rect2)

print(lib.cells)
# Save the library in a file called 'first.gds'.
lib.write_gds('hoge.gds')
```

I encountered an ValueError when I execute it more than twice.
This is because a cell instance is added to a `current_library` in default.

I would generate more than two gds files in a code in some cases, and these gds files should be independent. In that case, similar error will occur which is abovementioned.
In oder to avoid the error, I need to set a flag `exclude_from_current=True` or reset a `current_library` by `gdspy.current_library=gdspy.GdsLibrary()`.

However I think that when we use a library instance, current_library is not necessary in general.
Is my idea is wrong? If so, could you tell me your intention?

Thankyou

heitzmann, 2019-12-27:
@mshrn, you're right. Using `exclude_from_current=True` would be my choice in your case. This issue will be solved when `current_library` is properly removed from the module (it is currently deprecated).

ljstek, 2020-01-08:
@heitzmann I am having a similar issue when importing two distinct gds files which contain cells having the same name. Because of the way how the `gdspy.Cell` inserts itself into `gdspy.current_library` by default, this will result in cell name collisions. The expected behavior would be that the two gds files each maintain their own library and no cells from either gds file will be inserted into `gdspy.current_library`, as was the case previously in version 1.4. 

The following snippet should show you what I mean:
```python
import gdspy as gd

# Problem setup
rect_cell = gd.Cell('TOP', exclude_from_current = True)
rect_cell.add(gd.Rectangle((0, 0), (1, 1), layer = 0))

circ_cell = gd.Cell('TOP', exclude_from_current = True)
circ_cell.add(gd.Round((0, 0), 1, layer = 1))

gd.write_gds('rect.gds', cells = [rect_cell])
gd.write_gds('circ.gds', cells = [circ_cell])

# Check that current library is indeed empty
print(gd.current_library)

# Importing
rect_import = gd.GdsLibrary(infile = 'rect.gds')
print(gd.current_library) # Should be empty on v1.4, but is not on v1.5

circ_import = gd.GdsLibrary(infile = 'circ.gds')
# ^^^ Will produce error on v1.5 because of cell name collision in gd.current_library
# "Fallthrough" for v1.4:
print(gd.current_library) # Should be empty
```

I have solved this locally by changing [line 2348 in `library.py`](https://github.com/heitzmann/gdspy/blob/2f98a9ab51f7c73665a264fde71447262aa3a4db/gdspy/library.py#L2348) to the following:
```python
cell = Cell(name, exclude_from_current = True)
```
I understand this will not be an issue when `current_library` is removed in the future. However for the time being, I would suggest you change this in the master branch as well to prevent other people from being hit by the same issue.

heitzmann, 2020-01-08:
Thanks for the fix!

closed 2020-01-08 by heitzmann