heitzmann/gdspy #118: LayoutViewer: Group by datatype

McCrearyD, 2020-01-26:
Is it possible to have a "group by datatype" function within the layout viewer? Like say you have 5 layers 31-35 & each layer has a subgroup of rectangles that have 5 datatypes, say 1-5. The view is cluttered with 10 separate views, but it would be nice if you could organize them by having them grouped by their datatype. Like a drop down for datatype 0 and have the # of polygons with that datatype show up.

If this isn't already a feature, I could help by implementing it. If you could point me in the right direction as to where to implement that, I would be appreciative!

heitzmann, 2020-01-31:
@McCrearyD you can show/hide individual pairs of layer+datatype in the viewer by clicking over them. If I understand your question, you'd like to be able to show/hide based solely on datatype?
There should be something analogous for layer on that case, or, even better, some way of smart filtering based on both. If you want to give it a try, go ahead. It's been a while since I last edited LayoutViewer code, so it might be messy...

heitzmann, 2020-02-06:
@McCrearyD commit e8b7c3c should give a good way to filter layers/datatypes. You can, for example, leave only datatypes 2 and 5 visible (for any layer) by setting the filter to `(*/2,5)`.

closed 2020-08-02 by heitzmann