heitzmann/gdspy #121: read_gds read slowly

tkoyat, 2020-02-04:
Hi heitzmann:
Using  read_gds functions to read large files is extremely slow. Do you have a solution?
klayout read file speed is relatively fast

heitzmann, 2020-02-04:
There are many reasons why gdspy is slower, but mainly it's because it is written in python... it's hard to overcome that. Maybe there is a specific bottleneck we can improve. Could you send some profiling information for me to check which functions are the slowest?

tkoyat, 2020-02-04:
read BOX 0x2D....in a multi-process fashion, which I suspect will make it faster?

heitzmann, 2020-02-04:
Is your layout made only of boxes? Python doesn't multiprocess well, and even if that wasn't the case, most time would be spent in function calling, so it wouldn't matter.

heitzmann, 2020-08-02:
I've been working on a C++ library to use as a backend for gdspy 2.0, so I'll close this issue as unsolvable in the current versions.

closed 2020-08-02 by heitzmann