heitzmann/gdspy #122: Is there anyway to export vertex coordinates in gdspy

phylove01, 2020-02-24:
Hi,

I am looking for a way to export the vertex coordinates of gds files in python. I believe gdspy would be a good candidate to do it. Could you let me know how to export those values from gdspy. Or I would like to request to update that function unless the current version is possible.

Thanks,
Sooyong

heitzmann, 2020-02-26:
@phylove01 This is something that can be done more or less easily. You can load the library and use the [`get_polygons`](https://gdspy.readthedocs.io/en/stable/reference.html#gdspy.Cell.get_polygons) method on the top cell of the layout (or iterate over them, if there is more than 1 top cell).

This should give you a good starting point:
```python
import gdspy

lib = gdspy.GdsLibrary(infile="filename.gds")
for cell in lib.top_level():
    for coordinates in cell.get_polygons():
        # numpy array with dimension (N, 2) with all coordinates
        print("Coordinates:", coordinates)
```

closed 2020-08-02 by heitzmann

phylove01, 2020-08-03:
Thank you for your help! It was really helpful to develop a personal tool. I have one more question about your lib. I am gonna leave it on the other issue page. Thank you again!