heitzmann/gdspy #123: Remove label with specific name

vipinkypada, 2020-03-10:
Hi, 
I want to remove some labels with name "test<*>. Is it possible using gdspy.remove_labels command??

heitzmann, 2020-03-10:
Sure, maybe this will give you an idea on how to do that:
```python
def test(label):
    return label.text.startswith("test<") and label.text.endswith(">")

my_cell.remove_labels(test)
```

vipinkypada, 2020-03-11:
I'm getting the following error with this code.
AttributeError: 'str' object has no aatribute 'startswitch'.

And how can I remove certain labels without bus format. For example 'VDD'.

heitzmann, 2020-03-11:
You have a typo there... `startswith`, not `startswitch`.

vipinkypada, 2020-03-12:
Yes.. got it.. thank you... I've one more scenario, where I have multiple labels with test, test_1 and test_2. I need to delete only test label. This code deletes everything if I give startswith('test').  And please help me with the other attributes supported for the str object.

heitzmann, 2020-03-13:
In this second scenario you could simply change the test function to:
```python
def test(label):
    return label.text == "test"

my_cell.remove_labels(test)
```
or, even more compact, with a lambda function:
```python
my_cell.remove_labels(lambda label: label.text == "test")
```
As for other members and methods, you'd be better off checking out the Python documentation and tutorials.

vipinkypada, 2020-03-13:
It is working fine.. thanks a lot...

closed 2020-03-13 by vipinkypada