heitzmann/gdspy #124: Get coordinates of a path

bguttel, 2020-03-15:
Hi,

I was wondering if there is a function that will export the coordinates of a path in the shape of an array. 
Something similar to the get_polygons() function in the path classes, but one that will return the coordinates used to build this path.
Honestly I am not sure how to modify it for arcs, parametric curves etc., but at least in my case even the case of segments-only path will be great.

Thanks!

heitzmann, 2020-03-16:
@bguttel Not really. If you are using [`RobustPath`](https://gdspy.readthedocs.io/en/stable/reference.html#gdspy.RobustPath), you can call the object to get the coordinates of the central spine at specific points along the path:
```python
import numpy
import gdspy

path = gdspy.RobustPath((1, -1), 0.2)
path.segment((1, 1))
path.smooth([(-1, 1)], [numpy.pi / 2, -numpy.pi / 2], relative=False)
path.segment((-1, -1))

# The index 0 selects the 1st of the possibly many path in parallel described by this RobustPath
spine = numpy.array([path(u)[0] for u in numpy.linspace(0, len(path), 32)])

print(spine)
```
You can also try `get_polygons` on a `FlexPath` with zero-width, but the result will be duplicated (considering both sides of the zero-width polygon).

bguttel, 2020-03-16:
OK, this is not what I had in mind (I find 'FlexPath' to be more useful for my purposes) , but I will try to work with that.

Thank you for all the work you put into the package!

closed 2020-03-16 by bguttel