heitzmann/gdspy #127: How to use clipper.cpp directly from C++ script

LZYWJMHL, 2020-03-23:
Hi  Heitzmann:
    I have one problem bothering me for quite a while. How to use clipper.cpp directly from C++ script? Is there a similar function or class interface to use just like in the Python usage: clipper.clip(poly1, poly2, operation, 1 / precision), which is so simple and obsvious. It would be very nice if you could help me out. Great thanks in advance!

heitzmann, 2020-03-24:
@LZYWJMHL You probably want to look into the original library used in gdspy: http://angusj.com/delphi/clipper.php

The interface used by gdspy is not too different from the library itself, except, of course, for the data types.

LZYWJMHL, 2020-03-28:
> 
> 
> @LZYWJMHL You probably want to look into the original library used in gdspy: http://angusj.com/delphi/clipper.php
> 
> The interface used by gdspy is not too different from the library itself, except, of course, for the data types.

Hi @LZYWJMHL:
    Thank you so so much!!!

LZYWJMHL, 2020-06-02:
> 
> 
> @LZYWJMHL You probably want to look into the original library used in gdspy: http://angusj.com/delphi/clipper.php
> 
> The interface used by gdspy is not too different from the library itself, except, of course, for the data types.

Dear heitzmann:
    In the link you provided, I can't find the prototype of  DrawPolygons(). It can't be found in clipper. Could you show me where I can find such a function? Or do I have to write it myself? If so could you give me a hint how to write it? I am just a newcomer of self-taught programming language, and hope you could help me out a bit. Great thanks in advance! Have a nice day.

heitzmann, 2020-06-02:
@LZYWJMHL The library doesn't include a `DrawPolygons` function... I believe the examples in the front page are for an application the uses the library to do the operations and then presents the result.

closed 2020-08-02 by heitzmann