heitzmann/gdspy #128: Feature request: Add ability to get polygons on only single layer

sbuchbinder, 2020-03-31:
When dealing with large GDS files, using the `get_polygons(by_spec=True)` can use a lot of memory and take a long time.

Can adding a `on_layer=(layer, datatype)` optional parameter, which only adds polygons to the output dict if it is on the provided layer-purpose-pair, help reduce memory usage and speed up this function?

heitzmann, 2020-03-31:
@sbuchbinder Sounds reasonable. Maybe we can use the `by_spec` argument: `cell.get_polygons(by_spec=(layer, datatype))`. What do you think?

sbuchbinder, 2020-04-01:
This seems reasonable.

A check can be performed on the layer-purpose keys when in Cell.get_polygons and CellReference.get_polygons.
Should these checks also be performed for the get_polygons method for paths, or rely on the checks in the top-level Cell?

heitzmann, 2020-04-01:
I was thinking about that as well. I think there should be an "early check" to see if any of the path polygons have the desired (layer, datatype) specification to avoid the penalty of generating the polygons. Otherwise, generate all polygons and return only the desired ones. The other solution would be to only generate the desired polygons, but that increases the complexity of the methods a lot, so I'm more inclined to implement the simpler version.

heitzmann, 2020-04-01:
Please take a look at commit a1e46f66c6df430d2f3a46f027954135dd187194.
I believe it solves the issue.

sbuchbinder, 2020-04-02:
I did not get a chance to run a test (I am on a slightly older version of gdspy for compatability reasons), but the code looks good.

If I get a chance to test, and see any errors, I'll re-open.

Thanks for the speedy turn-around.

edmundsj, 2020-07-14:
I can confirm this works. I was just looking for this feature earlier!

closed 2020-08-02 by heitzmann