heitzmann/gdspy #129: Get Coordinates of FlexPath

drakesh307, 2020-03-31:
Hi ,
       I am a newbie to gdspy. I started using FlexPath recently . when i try getting the coordinates using get_polygons() function it gives me cordinates of outer boundary of the path . wondering if there a way i can get the coordinates of central spine for a flex path ?
Thanks

heitzmann, 2020-04-01:
@drakesh307 Somewhat related to #124
I'm not sure what it is you're trying to do, but there is also a [`Curve` class](https://gdspy.readthedocs.io/en/stable/gettingstarted.html#curves) that allows you to create open or closed curves to be used, for example, in creating polygons.

drakesh307, 2020-04-01:
hi @heitzmann , Thanks for a quick reply . I did look at issue #124 and tried to replicate the part of code you wrote to Robustpath for a flexpath but it didnt work. i am actually trying to export central spine coordinates of flexpath(with circular bend option enabled) to a different file and redraw my existing paths with new flexpath coordinates

heitzmann, 2021-01-22:
Unfortunately there's no automatic way of doing this. I'm closing it due to inactivity.

closed 2021-01-22 by heitzmann