heitzmann/gdspy #130: Problem inserting and referencing cell from a gdspy file to another.

fysikos6, 2020-04-11:
Hello everyone, thank you very much for this great module.
I am having the following issue which I cannot resolve.

I am trying to do the following:
I have a gdspy file which has a pattern inside it, and which I would like to programmatically place it in a bigger one ( basically it is a pattern of alignment markers, which I want to place in various places in a photomask). 
So I do the following as in the code, I make a new library, and in there I read the gds file. From this file I read the one and only existing cell. I then make two cell references, whichI add in a newly made cell. I then add this to a already existing library from before which contains the photomask.

However, when I try to check the saved gds file, I can see that there is an empty cell which I created but it has no content. Basically it is empty.  What could I be doing wrong in my case?

`
This is part of the script, here is the relevant part to the question.
markers_library=gdspy.GdsLibrary()
markers_library.read_gds(infile=r"C:\Users\fysikos6\Dropbox\phd\Mask_making\markers\MRidge.gds")
        
one_markers_cell=markers_library.extract(markers_library.top_level()[0])

test_library=gdspy.GdsLibrary()
ref0=gdspy.CellReference(one_markers_cell, (0,1000))
ref1=gdspy.CellReference(one_markers_cell, (0,3000))

cellforthenewlibrary=gdspy.Cell("malakasfilippos")

cellforthenewlibrary.add([ref1,ref0])

ridges_library.add(cellforthenewlibrary)

ridges_library.write_gds(r'C:\Users\fysikos6\Dropbox\phd\Mask_making\ridges.gds')`

heitzmann, 2020-04-11:
@fysikos6 Are you on the latest version of gdspy?

Note that `GdsLibrary.extarct` is deprecated. The better way to add the cell from one library to the other is simply using the `GdsLibrary.cells` dictionary or, in you case, using the top level cell directly.

The following should work on the latest gdspy release (1.5+):

```python
markers_library = gdspy.GdsLibrary(
    infile=r"C:\Users\fysikos6\Dropbox\phd\Mask_making\markers\MRidge.gds"
)
one_markers_cell = markers_library.top_level()[0]

# I assume this should be ridges_library, not test_library
ridges_library = gdspy.GdsLibrary()
ref0 = gdspy.CellReference(one_markers_cell, (0,1000))
ref1 = gdspy.CellReference(one_markers_cell, (0,3000))

cellforthenewlibrary = gdspy.Cell("malakasfilippos")
cellforthenewlibrary.add([ref1,ref0])

ridges_library.add(cellforthenewlibrary)

ridges_library.write_gds(r'C:\Users\fysikos6\Dropbox\phd\Mask_making\ridges.gds')
```

fysikos6, 2020-04-11:
Dear Lucas,
Thank you very much for the prompt answer.
I am using version 1.5.2
 I didn't know that `GdsLibrary.extract` was deprecated, I will refrain from using it further.
I tried to use the code you suggested, but I get the same result. The "ridges_library" library is actually an existing library from before, earlier in the script, in which I want to add the cell with the references.

A silly work around that gives the result I want is the following: 
`test_main=gdspy.GdsLibrary()
main = gdspy.Cell('Main')
main.add(ridges_cell)
main.add(cellforthenewlibrary)
test_main.add(main)
test_main.write_gds('test.gds')`
Where basically, I define a new library, and I add the cell with the references and the cell of the library I initially wanted to be my main (ridges_cell is included ridges_library)... It is a workaround that adds some unnecessary complexity, but I really can't figure what the problem is.

heitzmann, 2020-04-11:
I see... but I didn't find any mention of `ridges_cell` in your earlier snippet. Note that order matters in this case: when you add a cell A to a library, gdspy automatically includes any cells from references in A, but that won't work if the references are added latter.

The following code produces a file with a missing cell:
```python
cellA = gdspy.Cell("Main")
cellB = gdspy.Cell("Marks")

ref = gdspy.CellReference(cellB, (0, 0))

lib = gdspy.GdsLibrary()
lib.add(cellA)  # lib includes only "Main"
cellA.add(ref)  # lib does NOT change; it still includes only "Main"

lib.write_gds("cell_missing.gds")

# To address the problem, add cellB manually:
lib.add(cellB)
lib.write_gds("correct.gds")
```

This avoids the problem:
```python
cellA = gdspy.Cell("Main")
cellB = gdspy.Cell("Marks")

ref = gdspy.CellReference(cellB, (0, 0))
cellA.add(ref)

lib = gdspy.GdsLibrary()
lib.add(cellA)  # lib includes both "Main" and its dependency, "Marks"

lib.write_gds("correct.gds")
```

closed 2020-08-02 by heitzmann