heitzmann/gdspy #134: Memory leakaged caused by bounding box lookup

jayl940712, 2020-05-16:
Current get_bounding_box methods in /gdspy/library.py stores key for lookup in _bounding_boxes global dictionary. This causes object instance memory not able to release by garbage collector after calling get_bounding_box method. 

Since current_library is deprecated, I think it is best to remove _bounding_boxes lookup or clearly indicate what global variables to be cleared in the documentations.

Thanks.

heitzmann, 2020-05-18:
Makes sense! I've removed the global cache and left it only per cell. This will impact efficiency for references, but I don't want global states left after the next major update. (See commit 3978e4e)

jayl940712, 2020-05-18:
Thanks! Exactly what I meant!

closed 2020-05-18 by jayl940712