heitzmann/gdspy #135: FeatureRequest: Accept pathlib.Path (os.PathLike) as filepath input

gyger, 2020-05-25:
This is a small request also supporting the pathlib.Path (since python 3.4) or more general anything subclassing os.PathLike (since python 3.6) passing to the file methods.

e.g. in write_gds perhaps one could instead check if it has the write attribute if you still want to support python 2.7, otherwise checking for isinstance(f, io.IOBase) should work out.

        if isinstance(outfile, basestring):
            outfile = open(outfile, "wb")
            close = True
        else:
            close = False

Relevant references
- https://www.python.org/dev/peps/pep-0519/#os
- https://docs.python.org/3/library/os.html#os.PathLike

heitzmann, 2020-05-26:
I believe this check should work for all cases, right?
```python
close = True
if hasattr(outfile, "__fspath__"):  # Python 3.6+
    outfile = open(outfile.__fspath__(), "w")
elif isinstance(outfile, (basestring, pathlib.Path)):
    outfile = open(outfile, "w")
else:
    close = False
```

For python 2.7 we'd have to define `pathlib.Path` as a dummy class...

gyger, 2020-05-26:
I agree this should work. One could also not support pathlib on python 3.5 then the dummy class can be avoided.

```
close = True
if hasattr(outfile, "__fspath__"):  # Python 3.6+
    outfile = open(outfile.__fspath__(), "w")
elif isinstance(outfile, basestring):
    outfile = open(outfile, "w")
else:
    close = False
```

heitzmann, 2020-05-26:
Support added in commit 4c793fe

closed 2020-05-28 by gyger