heitzmann/gdspy #136: Translating a cell

vindi-pix, 2020-05-28:
Hi Heitzmann, 

I also really appreciate this great module. Lovely Work. Thanks you very much! 

In my project I also read loads of cells and place them with gdspy. The imported cells are not neccessary nicely centered around an meaningfull origin. Also they have several sub-references and loads of polygons atc. These Cells are then again referenced at different locations with different rotations. 

What is the best way to move all contents of a cell with out adding another hierarchical layer (Cell reference). Like centering them nicely before they get referenced again.  

Moving all contents of the cell (Polys, Labels, References etc.) sounds like a solution but is rather slow. 

Adding an intermediate cell and then flattening again removes hierarchy. 

Can you think of a nice solution ?

Best regards + Thank you

heitzmann, 2020-05-28:
Hi @vindi-pix,

Indeed translating every element or flattening a new cell would be very slow. Unfortunately there is no other option: it is either translating each element (polygon, path, label and references) or creating a new cell with a single reference at the desired position.

If I understand your issue, the problem is that your existing cells have strange origins, so when you place them in your layout, you have to inspect the cell you're referencing to find out how to place it, correct?

One possible solution for this case is to create a dictionary of corrections, so that when you reference a cell, the correction can be directly applied. Of course, to create the dictionary, you'll have to go through each possible cell you'll want to use and inspect it, but you're already doing it anyway. What I mean is something like this:

```python
correction = {
    "CELL1": numpy.array((some_x_pos, some_y_pos)),
    "CELL2": numpy.array((another_x, another_y)),
}

ref = gdspy.Reference("CELL1", origin=(desired_x, desired_y) + correction["CELL1"])
main_cell.add(ref)
```

vindi-pix, 2020-05-28:
Hi @heitzman, thanks for the quick answer. 

Yes, I also used the correction approch but then rotation is applied around original center. 
Or is this somehow parametric ? 

I mean it is possible to nicely center them before importing or shifting the content. I was just unsure, whether I did oversee this function ;)  

Best regards

heitzmann, 2020-05-28:
You're right, rotation is applied around the original center... you'll have to compensate for that...

vindi-pix, 2020-06-03:
I did. Everything worked out nicely. Thanks for your help + making gdspy ;) . Cheers

closed 2020-06-03 by vindi-pix