heitzmann/gdspy #137: UnicodeDecodeError when read a gds file converted from an oasis file

wylecho, 2020-06-19:
Hi @heitzmann ,

Thanks for your gdspy, I have been using it for a while.

It seems gdspy doesn't support oasis file read/write, so I used klayout to read in an oasis file and save it as gdsii format file. However, it still reported the following error when I try to read in the converted gds file:

> Traceback (most recent call last):
>   ... ...
>     lib= gdspy.GdsLibrary(infile=in_layout_file, unit=1e-6, precision=1e-9, layers={1}, datatypes={0})
>   File "/home/username/anaconda3/lib/python3.7/site-packages/gdspy/library.py", line 1989, in __init__
>     self.read_gds(infile, **kwargs)
>   File "/home/username/anaconda3/lib/python3.7/site-packages/gdspy/library.py", line 2272, in read_gds
>     for record in _record_reader(infile):
>   File "/home/username/anaconda3/lib/python3.7/site-packages/gdspy/gdsiiformat.py", line 96, in _record_reader
>     data = data.decode("ascii")
> UnicodeDecodeError: 'ascii' codec can't decode byte 0x8f in position 21: ordinal not in range(128)

I am not sure whether the converted gds file from klayout follows the gdsii format strictly. But it could be opened normally by klayout and other layout viewers. Do you have any ideas?

heitzmann, 2020-06-19:
This is probably due to some cell name (or the library name) containing non-ASCII characters. The official specification allows only ASCII characters for strings, but I don't see why we wouldn't be able to support utf-8 encoding (since it's ASCII-compatible).

If you change the line with the error to `data = data.decode("utf-8")` does it fix the problem?

wylecho, 2020-06-30:
@heitzmann , thanks for your reply.

I think there maybe something abnormal with my previous oasis file, I tried another oasis file and converted it into gdsii format, now it works well with gdspy.

closed 2020-06-30 by wylecho