heitzmann/gdspy #138: Is it possible to add a path with no width for MEEP simulation

zswhero, 2020-06-19:
Hi,

I am wondering if it is possible to write a path with no width (like only two points line with no width) for MEEP simulation. I have checked the gdspy.Path class. I find that the path will not display if I set the width to 0.

The code for the path with no width:
%~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
path_test = gdspy.Path(width=0, initial_point=(0, 0))
path_test.segment(2, "+y", layer=1)
%~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

heitzmann, 2020-06-19:
@zswhero, you can create it using the [`FlexPath`](https://gdspy.readthedocs.io/en/stable/reference.html#flexpath) class with `gdsii_path = True`:
```python
path = gdspy.FlexPath([(0, 0), (0, 2)], 0, layer=1, gdsii_path=True)
```
It correctly shows in the gds file. I'm not sure if meep will load it, though.

zswhero, 2020-06-20:
@heitzmann Yes. I test it and it works in MEEP. Thank you for your help.

closed 2020-06-20 by heitzmann