heitzmann/gdspy #141: Feature Request: Scaling/Translation/Rotation of Cells

edmundsj, 2020-07-14:
I am using gdspy to design a reticle for an ASML stepper tool, and I frequently need to scale and translate not just individual polygons, but an entire cell (for example the top-most cell in the hierarchy). The translate method is implemented for a CellReference, so I'm guessing internally all the guts are there to do this.

Why not just use the _magnification_ argument when writing the .GDS file? Because when designing my mask I need to merge it with a few other masks (a template for the fiducials and the barcode written and maintained by others). It would be great if the translate, scale, and rotate methods were implemented for cells. This would make my life much easier. 

Why do I need a translate argument for the top-most cell? Well, not really the *top-most cell*, but the next level down in the hierarchy. Because I have a multi-field mask, I need to split up my mask by layer, magnify those layers, move them into their appropriate fields, and then merge with the mask template (an existing .GDS file). Right now I'm just iterating through all the polygons and doing the translation and scaling manually.

heitzmann, 2020-07-15:
@edmundsj I understand what you're describing, but there is one issue to be aware of: the method that would transform a cell would, in fact, be transforming the elements of the cell. Any of those elements could be used by other cells and, as a consequence would be transformed there as well (this is not usual, but can be effective for memory-constrained layouts).
The solution would be to create a copy of all elements, transform those and substitute the cell elements. In the end it would be less efficient (memory and speed-wise) than your current solution.

edmundsj, 2020-07-16:
In pretty much everything I work on and my colleagues work on (academic setting with MEMS/photonics, not production with very complex masks) the memory hit would be more than worth it to get rid of the complexity involved in my hacky implementation (and I will likely forget that implementation in 6 months when I need to make another mask and have to figure this out all over again). Up to you, I see the trade-off. You could make some note in the docs as to the inefficiency of the methods and a suggested efficient alternative.

heitzmann, 2020-08-02:
@edmundsj Could you please check if this resolves your use case?

edmundsj, 2020-09-08:
Yup, this looks exactly like what I was doing, just crawling through all the objects and transforming them. Thanks for this! It will be very useful when I make my next mask.