heitzmann/gdspy #142: Length definition of polygon mask

aonurdasdemir, 2020-07-24:
Hi,
The package is very useful, thank you! I was wondering if there is a way to determine what will be the length scale of the mask easily. For example, I am creating a 3 um x 3 um shape with 450 nm input waveguide by drawing with polygons. However, when I import the GDS file, I see the length of the 450 nm waveguide can be like 200 um. After the import, I can measure the imported length and scale my gds file before importing but this would be a manual way to solve this problem. Do you have any suggestions on how I can scale my lengths? Maybe you can help me understand how the default lengths are determined by polygon points.

heitzmann, 2020-07-24:
@aonurdasdemir I'm not sure I understand your question... Can you give an example?
The length scales of your drawing are controlled by you. When you save the file, the argument `unit`defines the size of a unit in your design.

aonurdasdemir, 2020-07-24:
What do you mean by unit here?

heitzmann, 2020-07-24:
The unit you use to define you dimensions. When you create, for example, a square from (0, 0) to (1, 1), what is the length of each side? 1µm? 1nm? 1m? The argument `unit` defines what is the unit scale (in meters). If you set `unit = 1e-6`, then the square size if 1µm. Please take a look at [the explanation in the docs](https://gdspy.readthedocs.io/en/stable/gettingstarted.html#units-in-gdsii).

closed 2021-01-22 by heitzmann