heitzmann/gdspy #143: How to implement other GDS dialects?

gyger, 2020-08-02:
Unfortunately there is this bad practice of industry people to implement there own GDS dialects (in my case RAITH).

If I want to add this elements to the library, do you think monkey patching is the way to go?

heitzmann, 2020-08-02:
Hi @gyger, what do you mean by "dialect"? Are these different elements?
If it is something we can implement without changing the default behavior, I'm open to it.

gyger, 2020-08-02:
That is of course even better to hear. There are two new record tags in the GDS file.

```
    'RAITHCIRCLE': 0x5600,
    'RAITHFBMS': 0x5800,
```

The circle element, is quite well captured by the gdspy.polygon.Round element. It can also describe ellipses, rings, and parts of that.

The FBMS (Fixed beam moving stage) is probably best described by a Path (it can also describe turns with angles)

There is one "reference" library for the [RAITH GDS extension](https://github.com/nrc-cnrc/Raith_GDSII) available in MATLAB.  That is MPL license, so should be nice from this perspective.

Another person implemented, parts in the [python-gdsii](https://github.com/cdoolin/python-gdsii-raith) library by forking it.

heitzmann, 2021-01-22:
@gyger I must have missed your reply here, sorry.
In any case, Gdspy is feature-frozen now, so this proposal would have to be implemented in [Gdstk](https://github.com/heitzmann/gdstk/).

closed 2021-01-22 by heitzmann

userx14, 2023-04-05:
Hi,

currently I'm working on including Raith elements in a fork of gdspy:
https://github.com/userx14/gdspy
Still WIP, but someone might find it useful.

I couldn't get [python-gdsii](https://github.com/cdoolin/python-gdsii-raith) to work, I think that their code for the FBMScircle might work, but the code for FBMS paths does not match the Raith matlab library.

Best,
Benjamin