heitzmann/gdspy #144: Plz let me know to translate gds to image.

phylove01, 2020-08-03:
Hi, 

I am trying to translate a gds clip to an image. I couldn't find out the sub function of lib, "write_svg" on the ver. 1.5.2. unlike your instruction on the web page. Could you let me know how to translate the gds to th image on your library.

heitzmann, 2020-08-03:
Hi @phylove01, the [`write_svg`](https://gdspy.readthedocs.io/en/stable/reference.html#gdspy.Cell.write_svg) method is part of `Cell`, not `GdsLibrary`; maybe that's why you didn't find it. Let me know if you need any help with it.

phylove01, 2020-08-05:
Thank you for your help! I have one more question. Is there anyway to export the rasterized image of gds like png, jpg instead of the vector image like svg format.

heitzmann, 2020-08-05:
Not directly, but you can use Inkscape to do the conversion (it has command-line arguments to use it directly without the interface).

microchipster, 2020-09-18:
Hello,

I was looking to use this library for the same thing, to render a gds as a png. In case it is useful or can get added to the main code, I just wanted to report that when using `<cell>.write_svg(...)` I've had to do the following to post process the svg prior to converting to png:

```python
from gdspy import GdsLibrary

def patch_svg(input_path: Path, output_path: Path) -> None:
    """ SVG does not allow < and & symbols between html style tags. Escape these correctly. """
    output_lines = []
    for line in input_path.read_text().split("\n"):
        new_line = line.replace(">& ", "> ")
        match = re.match(r"(.*)>([A-Za-z0-9_-]+)<(\d+)>(.*)<(.*)", new_line)
        out_line = str(new_line)
        if match:
            groups = match.groups()
            out_line = f"{groups[0]}>{groups[1]}&lt;{groups[2]}&gt;{groups[3]}<{groups[4]}"

        output_lines.append(out_line.replace(">&", ""))

    output_path.write_text("\n".join(output_lines))

def gds_to_svg(input_gds: Path, output_svg: Path, style_dict: t.Dict) -> None:
    gdsii = GdsLibrary(infile=str(input_gds))
    top_cells = gdsii.top_level()[0]
    top_cell.write_svg(str(output_svg), style=style_dict)
```
I imagine there might be a better way to handle the html character encodings (e.g. `>` -> `&lt;`) internally though.
 
Then to convert svg to png, I've found the fastest and most reliable way is using cairosvg `pip install CairoSVG` (had some issues with getting imagemagick to behave as expected, and doing it in pure python was important for my use case):
```
from cairosvg import svg2png

svg2png(url=str(input_svg), write_to=str(output_png))
```

My goal was to replace a flow I typically go through with something that could be automated and not require a proprietary license. I would normally use a proprietary tool or klayout (unfortunately unavailable for me in the execution environment) to manually view and screenshot the layout for use in presentations or to sanity check the layout.

With the patch above, this is working great (thanks for this cool lib!), but the one issue I'm having now is that it is extremely slow if I run on a large design. This might be a fundamental limitation, but when opening in klayout for example it renders quickly on the screen, so I was wondering if there are any tips or modifications that could be made under the hood for speeding it up if high resolution rendering can be sacrificed?

heitzmann, 2020-09-18:
@microchipster Thanks for the suggestions! I didn't really think of sanitizing the strings that go into the SVG, but it can certainly be a problem. It should be more or less doable within the library itself.

The speed problem is the one I'm most bothered with, that's why I've been working on [Gdstk](https://github.com/heitzmann/gdstk), the [successor to Gdspy](https://github.com/heitzmann/gdspy#future-of-gdspy). It is still unreleased, because I'm writing tests and fixing bugs as they surface, but it is already promising.

heitzmann, 2020-09-18:
@microchipster Where exactly did you find the "<" character between HTML tags? Can you share a layout or SVG with the issue?

microchipster, 2020-09-18:
It's in the text layers of the gds. The pin indices are denoted with <0> etc.
So the raw svg out of gdspy has entries like this:
```html
<text class="l132t0" text-anchor="end" dominant-baseline="central" transform="scale(1 -1) translate(258.2 -123.85) rotate(-90.0) scale(0.05)">PIN_NAME<1></text>
<text class="l132t0" text-anchor="end" dominant-baseline="central" transform="scale(1 -1) translate(249.85 -123.85) rotate(-90.0) scale(0.05)">PIN_NAME<2></text>
<text class="l132t0" text-anchor="end" dominant-baseline="central" transform="scale(1 -1) translate(248.45 -123.85) rotate(-90.0) scale(0.05)">PIN_NAME<3></text>
```

Right after I posted I noticed the comments about gdstk in the readme, looks exciting, will give that a try!

closed 2020-09-23 by heitzmann