heitzmann/gdspy #145: why different eight byte real data has same double value

richardxu80, 2020-08-04:
I tried to convert a random eight byte array to float64, and then convert the double data to eight byte array.
but can't pass the test, because different eight byte array can convert to same double value. is there any wrong of it?
`def test_twoway():
    f = gdspy.gdsiiformat._eight_byte_real_to_float
    g = gdspy.gdsiiformat._eight_byte_real
    for x in [0, 1.5, -numpy.pi, 1 / 3.0e12, -1.0e12 / 7, 1.1e75, -0.9e-78]:
        assert x == f(g(x))
    for _ in range(10000):
        x = numpy.random.bytes(8)
        assert x == g(f(x))`

heitzmann, 2020-08-04:
@richardxu80 I believe this is because not all GDSII reals can be exactly represented as floats (double) and vice-versa. I'm guessing the round-trip causes some loss of precision. The following test shows that the error is almost always in the least significant byte:

```python
err_count = 0
soft_err_count = 0
for _ in range(10000): 
    x = numpy.random.bytes(8)
    y = g(f(x)) 
    if y != x: 
        err_count += 1 
        print(x.hex(), y.hex(), f(x))
        if y[:-1] != x[:-1]:
            soft_err_count += 1
print(err_count, soft_err_count)
```

I get about 75% error, but only 7% beyond the last byte. There are instances of "argument out of range" errors, so I'll investigate a little further. If you find anything else, let me know.

heitzmann, 2020-08-04:
@richardxu80 There are 2 issues with the random test:
* not all random 8-byte sequences represent valid numbers in the GDSII 8-byte real format. In particular the mantissa needs some normalization. In the end, you have to guarantee that the second most significant byte is at least 16
* The transformation between formats is not exact, so there is some loss in the round-trip conversion. The following test should clarify what I mean:
```python
import numpy
import gdspy

f = gdspy.gdsiiformat._eight_byte_real_to_float
g = gdspy.gdsiiformat._eight_byte_real

real_err_count = 0
float_err_count = 0
count = 1000000
for _ in range(count):
    r1 = numpy.random.bytes(8)
    while r1[1] < 16:  # Guarantee normalization
        r1 = numpy.random.bytes(8)
    f1 = f(r1)
    r2 = g(f1)
    f2 = f(r2)
    if r1 != r2:
        real_err_count += 1
    if f1 != f2:
        float_err_count += 1

print(real_err_count, float_err_count)
```
Errors in the real representation are around 73%, but there are *no float errors*, which means that `r1` cannot always be exactly represented by `f1`.

closed 2020-09-23 by heitzmann