heitzmann/gdspy #146: Circular import error with gdspy with clipper module

francesco-st, 2020-08-05:
Hi all,

I was trying to install and use gdspy on windows. I have python 3.8.1. I have tried to install the wheel for gdspy from the release page. I have previously ensured that numpy was installed and functional.

When I test using "import gdspy" I get the following error:

```
Python 3.8.1 (default, Dec 24 2019, 12:23:31)  [GCC 9.2.0 32 bit] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import gdspy
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:/francesco/local/lib/python3.8/gdspy/__init__.py", line 26, in <module>
    from gdspy.library import (
  File "C:/francesco/local/lib/python3.8/gdspy/library.py", line 42, in <module>
    from gdspy.polygon import PolygonSet, Polygon
  File "C:/francesco/local/lib/python3.8/gdspy/polygon.py", line 39, in <module>
    from gdspy import clipper
ImportError: cannot import name 'clipper' from partially initialized module 'gdspy' (most likely due to a circular import) (C:/francesco/local/lib/python3.8/gdspy/__init__.py)
```

Now I recognize that my install environment is odd because I am on Windows using msys2. python and numpy are provided by msys2 and they works. gdspy was installed using the wheel file from the release page.

In my understanding this kind of error is not related to Windows or to the way I was trying to install but is a problem in the logic of imports.

I would need to use gdspy to get some work done :-) please help me fix this problem.

I may try to compile gdspy from the source if needed. I am used to compile applications or libraries when needed.

Thank you in advance for your amazing work and for any help you may provide.

Francesco

francesco-st, 2020-08-05:
I have just tried to compile and install from the source code. The build and install completes successfully but when I try to actually import gdspy I get exactly the same error about circular import with clipper.

heitzmann, 2020-08-05:
@francesco-st It's strange because I'm not sure there are circular imports in gdspy. `clipper` in particular is only imported there, IIRC. It could also be that the compiled library cannot be imported because of some other error, e.g., different compiler versions (that's why the wheel did not work — it is compiled with MSVC). I see your Python has been compiled with GCC. How did you compile gdspy? Can you share the full terminal session?

franko, 2020-08-05:
Hi @heitzmann,

same person here, I am just using my personal account. I have just tested on Linux and I have almost the same error, without a mention of the circular import. My programmer spider sense tell me it is a circular reference:

during the import of gdspy, inside polygon.py there is a call `import gdspy from clipper`. The problem is that while it is importing gdspy, polygon ask to load something from gdspy. To me it looks like is a circular import problem.

By the way, here the details how what I do on linux:
```sh
# install numpy for python 3
sudo apt install python3-numpy

# clone gdspy repository and enter into the directory
git clone https://github.com/heitzmann/gdspy.git && cd gdspy

# build and install
python3 setup.py install --prefix $HOME/.local
```
The build and install is successful, here the [output of the install command](https://github.com/heitzmann/gdspy/files/5030064/gdspy-setup-install-log.txt).

Then, when I try to `import gdspy` I get the following error:

```
francesco:~/dev/gdspy$ python3
Python 3.6.9 (default, Jul 17 2020, 12:50:27) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gdspy
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/francesco/dev/gdspy/gdspy/__init__.py", line 26, in <module>
    from gdspy.library import (
  File "/home/francesco/dev/gdspy/gdspy/library.py", line 42, in <module>
    from gdspy.polygon import PolygonSet, Polygon
  File "/home/francesco/dev/gdspy/gdspy/polygon.py", line 39, in <module>
    from gdspy import clipper
ImportError: cannot import name 'clipper'
```

heitzmann, 2020-08-05:
There's something weird, though: your installation log says it installs the package in `.local`, but the error indicates that it is trying to find the module in `dev`.
I believe the issue is that you're running from inside the gdspy directory, as your path indicates, so python is trying to import the module from the source tree (where there is no compiled `clipper`) instead of the installation path.
Try to run python3 from your home directory... it should work.

franko, 2020-08-05:
Oh, yes you are right! From the work directory it is working. Weird problem.

Thank you very much for your help. You can close the issue of course.

closed 2020-08-05 by heitzmann