heitzmann/gdspy #147: Adding and relocating a gds design to another design

kagankoral, 2020-08-18:
Hello,
Thank you for this useful package. I was wondering that, I have a gds file and this gds file consists of many cells. I want to transfer this gds design to a new design. Also, when I export this file, I want to change the location of this multi-cell design. When I do this process myself, I do it by entering all the cells one by one and changing the coordinates. Is there an easier way to get this process done without entering the cells?

heitzmann, 2020-08-18:
@kagankoral Can you share a your code (or a small piece of it) so I can better understand the problem?

Do you need to move the contents of the cells? Or do you have a main cell with many references and you need to move the references? And are they all translated by the same amount?

kagankoral, 2020-08-19:
`import gdspy

lib = gdspy.GdsLibrary()

lib.read_gds('multi_layer.gds')
polygon_cell = lib.new_cell('POLYGON')
cells = lib.cells

obj1 = cells.get('LAYER1')
object1 = obj1.polygons[0]

obj2 = cells.get('LAYER2')
object2 = obj2.polygons[0]

obj3 = cells.get('LAYER3')
object3 = obj3.polygons[0]

points = [(10,10),(30,10),(20,0)]
polygon_obj = gdspy.Polygon(points, layer=10, datatype=0)
obj3.add(polygon_obj)

for i in range(0,4):
    for j in range(0,2):
        if (i==0 and j==0):
            object3.polygons[0][i][j] = 10
            object2.polygons[0][i][j] = 12.5
            object1.polygons[0][i][j] = 15
        if (i==0 and j==1):
            object3.polygons[0][i][j] = 10
            object2.polygons[0][i][j] = 12.5
            object1.polygons[0][i][j] = 15
        if (i==1 and j==0):
            object3.polygons[0][i][j] = 10
            object2.polygons[0][i][j] = 12.5
            object1.polygons[0][i][j] = 15
        if (i==1 and j==1):
            object3.polygons[0][i][j] = 30
            object2.polygons[0][i][j] = 27.5
            object1.polygons[0][i][j] = 25
        if (i==2 and j==0):
            object3.polygons[0][i][j] = 30
            object2.polygons[0][i][j] = 27.5
            object1.polygons[0][i][j] = 25
        if (i==2 and j==1):
            object3.polygons[0][i][j] = 30
            object2.polygons[0][i][j] = 27.5
            object1.polygons[0][i][j] = 25
        if (i==3 and j==0):
            object3.polygons[0][i][j] = 30
            object2.polygons[0][i][j] = 27.5
            object1.polygons[0][i][j] = 25
        if (i==3 and j==1):
            object3.polygons[0][i][j] = 10
            object2.polygons[0][i][j] = 12.5
            object1.polygons[0][i][j] = 15
obj3.add(object3)
obj2.add(object2)
obj1.add(object1)

lib.write_gds('new_object.gds')`

This is the code I wrote.  I want to add the multi-layer object to the polygon that I created. You can find this multi-layer object in the following zip file. As you can see in the code above, I do this by entering each cell and changing the coordinates of the polygon in these cells.
[multi_layer.zip](https://github.com/heitzmann/gdspy/files/5097690/multi_layer.zip)

heitzmann, 2020-08-20:
@kagankoral I see you are changes each vertex of almost all polygons, one by one. You don't need a loop for that, you could simply do, for example,
```python
object3.polygons[0] = numpy.array([10, 10], [10, 30], [30, 30], [30, 10]])
```
but at this point it is simply easier to recreate the polygons and cells themselves:
```python
lib = gdspy.GdsLibrary()

cell3 = lib.new_cell("LAYER3")
poly3 = gdspy.Rectangle((10, 10), (30, 30), layer=3, datatype=3)
cell3.add(poly)

# ... same for other shapes and cells ...

lib.write_gds("new_object.gds")
```
and so on.

If you need to move more complex polygons, use, for example, [`object3.translate(20, 20)`](https://gdspy.readthedocs.io/en/stable/reference.html#gdspy.PolygonSet.translate).

closed 2021-01-22 by heitzmann