heitzmann/gdspy #150: Draw lower hierarchy layer in top level

vipinkypada, 2020-08-27:
Hi,

I'm trying to find the coordinates of a layer(15:0) from lower hierarchy and draw it in the top hierarchy exactly on top of it. Is there a direct command to do it considering the origin and orientation of the bottom hierarchy cells(5 levels down). Can I use the get_polygons command and transpose the lower coordinates to the top?  Could you please help me to find out a solution for this.

Thank you
Vipin

heitzmann, 2020-08-27:
@vipinkypada The method `get_polygon()` with `by_spec=True` is the best solution. The polygon you're looking for is in layer 15 and datatype 0, is that correct?
```python
all_polygons = top_reference.get_polygons(True)
possible_polygons = all_polygons[(15, 0)]
```
If you have more than a single polygon in layer/datatype (15, 0), you'll have to find witch one in the list `possible_polygons` is the one you want, but that should give you an idea of the solution.

vipinkypada, 2020-08-28:
Hello Heitzmann,

Thanks for the quick response. The above mentioned code works fine for all the layer15:0 in top level. I'm looking for the layers from the bottom hierarchy cells. I've 10-15 cells in 3 hierarchy down. The origin and rotation is different for all of them. How can I transpose the coordinates of layer15:0 from these cells and duplicate them in top level by considering these origins and rotations.
Thanks
Vipin

heitzmann, 2020-08-28:
The previous solution will get polygons from all cells down the hierarchy. I'm bot sure what you mean by working only.in the top level. Can you share your code and a sample gdsii file?

closed 2021-01-22 by heitzmann