heitzmann/gdspy #151: GDS creation take time for 500M rectangles

SatyajitKautkar, 2020-09-23:
Hi,

I am trying to generate GDS file with 5 layers, with each layer having 100 millions rectangles.
It is taking more than 10 hours to generate and write the file.  Is is expected?
I would say most of the time is consumed in writing the gds file.
Could you please let me know how to optimize it if it is possible?

Below is my code:

```
GDSCELL=gdspy.Cell("Topcell")
for j in range (0, 100000000): 
        x= random.randint(0,10000000)
        y= random.randint(0,10000000)
        q= random.randint(10,100)
        k= random.randint(10,100)
        GDSCELL.add(gdspy.Rectangle((x,y),((x+k),(y+q)),layer,1))    
gdspy.write_gds('Test_500M.gds',unit=1.0e-9,precision=1.0e-9)
```
`

heitzmann, 2020-09-23:
@SatyajitKautkar This is the expected behavior, unfortunately. Python has a lot of overhead in function calls, and to write the gds file in the last step we have to call the to_gds method of each polygon. Multi-threading will have little impact there. Your best bet is to move over to [Gdstk](https://github.com/heitzmann/gdstk). I ran a quick test on my notebook and got the following wall time results:

| Module | num = 1000 | 10000 | 100000 | 1000000 | 10000000 |
| :--------- | --- | --- | --- | --- | --- |
| Gdspy | 0m0.247s | 0m0.524s | 0m3.573s | 0m37.370s | 5m37.081s |
| Gdstk | 0m0.213s | 0m0.253s | 0m0.964s | 0m6.073s | 1m12.345s |

Script for Gdspy:
```python
import sys
import random
import gdspy

num = int(sys.argv[1])
gdscell = gdspy.Cell("Topcell")
for j in range(0, num):
    x = random.randint(0, 10000000)
    y = random.randint(0, 10000000)
    q = random.randint(10, 100)
    k = random.randint(10, 100)
    gdscell.add(gdspy.Rectangle((x, y), ((x + k), (y + q)), 1, 1))
gdspy.write_gds("Test_500M.gds", unit=1.0e-9, precision=1.0e-9)
```
and for Gdstk:
```python
import sys
import random
import gdstk

num = int(sys.argv[1])
lib = gdstk.Library(unit=1.0e-9, precision=1.0e-9)
gdscell = lib.new_cell("Topcell")
for j in range(0, num):
    x = random.randint(0, 10000000)
    y = random.randint(0, 10000000)
    q = random.randint(10, 100)
    k = random.randint(10, 100)
    gdscell.add(gdstk.rectangle((x, y), ((x + k), (y + q)), 1, 1))
lib.write_gds("Test_500M.gds")
```

SatyajitKautkar, 2020-09-23:
Thank you for the inputs. I really appreciate it.
I will try using gdstk for generating the same set of layers.

closed 2020-09-23 by SatyajitKautkar