heitzmann/gdspy #152: Square array boolean time

YangXu024, 2020-10-05:
Hi,

I want to do 'not' boolean for a 500*500 number of square arrays with a special shape. But it takes several hours to do such boolean. (I mean substract a special shape out from 2500 square background). Is there a way to reduce the calculation time?

Best,
Yang

heitzmann, 2020-10-06:
@YangXu024 The boolean function is written as a C-extension already, so I'd say it is already as fast as it could be. You could try [Gdstk](https://github.com/heitzmann/gdstk) to see if it somehow improves speed, but my first guess would be no (or very little).

You could try to decrease the size of your problem, if possible. For example, does the shape you're subtracting intersect all 250.000 squares? Maybe you can use [`gdspy.inside`](https://gdspy.readthedocs.io/en/stable/reference.html#gdspy.inside) to filter out some squares before the subtraction. How is the number of vertices in this shape? Boolean operation time increases fast with the number of vertices, so, if you can simplify all shapes, that will also help.

YangXu024, 2020-10-11:
Hi, thanks a lot, gdspy.inside really helps! BTW, may I ask is there an easy way to increase all components 4 times in scale?

Best,
Yang Xu

YangXu024, 2020-10-11:
Hi, thanks a lot. gdspy.inside really helps. BTW, when I load another gdsII contents in existing library, existing contents disappear(I can see layers but contents is not shown) and only the new loaded contents are shown. May I ask what might be the reason? I just use lib.read_gds('contents.gds').

heitzmann, 2020-10-12:
@YangXu024 It could be a matter of scale. Does the library use the same units as the gds file? Maybe try `units="convert"` in [`GdsLibrary.read_gds`](https://gdspy.readthedocs.io/en/stable/reference.html#gdspy.GdsLibrary.read_gds) to see if it helps.

YangXu024, 2020-10-13:
[image]
Hi, Thanks for the advise. It seems unit is not the problem after I tried convert. I attached a picture a my layer after read a new gdsII file. First 5 layers are from the new file, while the other grey layers are the old ones. I guess grey means it cannot be shown. 

Best,
Yang

YangXu024, 2020-10-13:
Sorry to bother you many times, may I ask is it possible to repeat only specific layers in a cell to get some array?

YangXu024, 2020-10-14:
Hello, 
Thanks a lot. You can check my code and file at https://github.com/YangXu024/twpa. The codes are a little bit messy... basically I creat a cell called Jline and want to add a mask from another library ASML 500-300 to it.

heitzmann, 2020-10-14:
I believe your GDS contains all the correct geometry; the problem might which cell you're looking at. In KLayout, did you select "main" to be the top cell (right click and show as net top)?

YangXu024, 2020-10-14:
Hello, 
Really thanks for your patience. With your suggestion I understand the issue now. It is that "main" cell and loaded cell are in the same hierarchy, such that they cannot be shown at the same time. 

I tried to put loaded cell under "main" cell. The thing is when a cell "A" (e.g.) in lib 1 is added to lib 2, the lower hierarchy cells in "A" won't be copied to lib 2.

YangXu024, 2020-10-14:
Sure, I solve the problem with your suggestion, thanks a lot. Just mention that the loaded cell and old cell are in the same hierarchy and they cannot be shown at the some time in Klayout (as I see). Also, when a cell "A" in lib 1 is added to lib 2, lower hierarchy cells in "A" wont be copied to lib 2, maybe it's some inherit problem for class. Anyway, thank you so much.

heitzmann, 2020-10-14:
Yes, when you want to include a cell from another library, you have to manually include all dependencies as well. I'm glad it worked out.

YangXu024, 2020-10-18:
Hi, is it normal that after boolean there are many additional lines, epecially when the points are very dense?

heitzmann, 2020-10-19:
I wouldn't say "normal", but it can happen depending on how irregular the geometry is and the precision you're using.

closed 2021-01-22 by heitzmann