heitzmann/gdspy #153: Best way to construct large repetitive patterns

stormyweathers, 2020-10-15:
Hey there,

Thanks so much for putting this project out there.  Its been a lifesaver.
I'm using gdspy for making diffraction gratings.
My patterns consist of large (mostly) periodic tilings of things like stripes, squares,triangles.
I'd like to generate patterns with 10^8-10^9 tiles.

I've tried to hierarchically generate cell references to build larger tiled areas, but this approach seems limited.
For example, on triangular cell it takes 106s to generate 10^6 tiles and takes 600s to generate 2x10^6 tiles.

Do you have suggestions about how to approach this?

I don't think gdstk will fix these orders of magnitudes out of the box, so I probably need to rethink my strategy.

heitzmann, 2020-10-15:
Hi @stormyweathers, these are indeed quite large numbers... I'm guessing you're already using `CellArray` everywhere it is possible, right? Other than that, it's hard to say what could be improved without looking at an example at least.

I agree that the Python interface for Gdstk might not help completely (although it is certainly faster), but if you need the best performance, you should think about using the C++ library directly. But be aware that the API is definitely prone to changes, so your program will probably break with newer versions.

stormyweathers, 2020-10-19:
Hey lucas,

For a rectangular latttice, i use `CellArray` to generate a 128x128 unit
cell and then recursively construct larger lattices from that.  I can
generate ~10^8 rectanges in 1 minute this way.  This method is faster than
using `CellArray` on the full 10^8 lattice.

For the triangular lattice i am not using `CellArray` because i haven't
found a way to build a large unit cell and recursively make a larger
lattice since the lattice vectors are at 60 degrees in this cell.

I will post an example later today

heitzmann, 2020-10-19:
@stormyweathers Please show the example, because I don't understand why it takes so long. Here is an example with 2 arrays, 1 rectangular and 1 triangular, each with 10.000 × 10.000 = 10⁸ elements. Layout generation is instantaneous and the GDSII file has only 528 B:

```python
import gdspy

num = 10000
d = 0.4

unit_cell = gdspy.Cell("Unit", exclude_from_current=True)
unit_cell.add(gdspy.Round((0, 0), d / 4, tolerance=d / 200))

rect_cell = gdspy.Cell("Rectangular", exclude_from_current=True)
rect_cell.add(gdspy.CellArray(unit_cell, num, num, (d, d)))

tri_cell = gdspy.Cell("Triangular", exclude_from_current=True)
tri_cell.add(gdspy.CellArray(unit_cell, num, num // 2, (d, d * 3 ** 0.5)))
tri_cell.add(
    gdspy.CellArray(
        unit_cell, num, num // 2, (d, d * 3 ** 0.5), (d / 2, d / 2 * 3 ** 0.5)
    )
)

lib = gdspy.GdsLibrary()
lib.add(rect_cell)
lib.add(tri_cell)
lib.write_gds("pattern.gds")
```

Note that if you need more than 30.000 elements per array dimension, then you'll have to build it in hierarchically (32.767 is the maximal number of columns or rows in a `gdspy.CellArray`).

stormyweathers, 2020-10-26:
Thanks for responding!  You've cleared up 2 issues for me.

1) my flat examples using `gdspy.CellArray` were stalling because I was attempting to view them with `gdspy.LayoutViewer()`.
This rendering operation is very slow for large arrays.  My computer takes at least several minutes to open your example.  Its already too laggy to be workable with `num=1000` on my computer.

2) It wasnt clear to me from the api reference that I could do non-rectilinear 2d grid with CellArray.  Thats way easier than the garbage I was doing.

I haven't gone past 2**15 element per array dimension yet, but thats good to know that the limit is close by.

So it seems that my problems are all solved, if I don't want to view the results.
I never need to see the full array, but I've been using `gdspy.LayoutViewer()` for validating my full-wafer layouts.

Do you have a suggested workaround?

heitzmann, 2020-10-26:
Glad it is working now! Indeed LayoutViewer is only for the most basic layouts. I don't really use it anymore. My experience is that KLayout works pretty well even for large layouts, but I don't think I've ever tried something on the scale you're talking about!

As for the CellArray, it is only rectangular... What I did is to interleave 2 rectangular arrays to produce the triangular one.

stormyweathers, 2020-10-26:
Ok cool! KLayout works for these giant patterns up to number=32767 and is very responsive when scrolling around.

I'm making diffraction gratings for NIR visible light.  For my dimensions, this will let me make gratings that are ~9 cm^2 without messing around with hierarchical construction.

Thanks a lot for your help!

closed 2020-10-30 by stormyweathers