heitzmann/gdspy #154: Loading gds file from LayoutEditor, got unexpected keyword argument 'width_transform'

jmb-grip, 2020-10-22:
I just installed gdspy from the most recent python 3.8 wheel. I can load gds files that were generated by gdspy, but when I load a file generated by the software LayoutEditor, the below error comes up. It seems that the Label class is getting passed `width_transform` even though it doesn't accept this kwarg.

Is this an issue with my file or a bug with gdspy?

```python
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-6-1ac7af44c8b0> in <module>
----> 1 lib_loaded = gdspy.GdsLibrary(infile='test.gds')

~\miniconda3\envs\gds\lib\site-packages\gdspy\library.py in __init__(self, name, infile, unit, precision, **kwargs)
   2179         self.precision = precision
   2180         if infile is not None:
-> 2181             self.read_gds(infile, **kwargs)
   2182 
   2183     def __str__(self):

~\miniconda3\envs\gds\lib\site-packages\gdspy\library.py in read_gds(self, infile, units, rename, rename_template, layers, datatypes, texttypes)
   2492             elif record[0] == 0x11:
   2493                 if create_element is not None:
-> 2494                     el = create_element(**kwargs)
   2495                     if len(properties) > 0:
   2496                         el.properties = properties

~\miniconda3\envs\gds\lib\site-packages\gdspy\library.py in _create_label(self, xy, width, ends, **kwargs)
   2636     def _create_label(self, xy, width=None, ends=None, **kwargs):
   2637         kwargs["position"] = xy
-> 2638         return Label(**kwargs)
   2639 
   2640     def _create_reference(self, **kwargs):

TypeError: __init__() got an unexpected keyword argument 'width_transform'
```

heitzmann, 2020-10-22:
Indeed that's a bug! I've just patched the source in the master branch. If you can, try it out and let me know if it works.

jmb-grip, 2020-10-23:
That seems to fix it! My file loads correctly now.

I'm a bit surprised your fix totally works though; shouldn't you have to change what kwargs are accepted by __init__ in the Label class?

heitzmann, 2020-10-23:
That argument is not supported by the Label class. AFAIK it has no meaning for a label in the GDSII specification, so the fix just ignores the argument.

It's strange to me that the software you used actually set that flag. I'd never seen that before.

closed 2021-01-22 by heitzmann