heitzmann/gdspy #156: simplify installation with conda package

joamatab, 2020-11-10:
Hi Lucas,

some windows users have reported that is hard to install gdspy on windows

I proposed them 2 options:

1. Install the wheel for the python version that they have, for example for python 3.7 or 3.8:

```
pip install https://github.com/heitzmann/gdspy/releases/download/v1.6.1/gdspy-1.6.1-cp37-cp37m-win_amd64.whl
```
```
pip install https://github.com/heitzmann/gdspy/releases/download/v1.6.1/gdspy-1.6.1-cp38-cp38-win_amd64.whl
```

2. Install a C++ compiler ["Build Tools for Visual Studio"](https://www.visualstudio.com/downloads/#build-tools-for-visual-studio-2017)

A third option would be to build a conda package

how could we build a conda package, so they could install gdspy with `conda install gdspy`?

heitzmann, 2020-11-10:
It is already available: https://anaconda.org/conda-forge/gdspy

joamatab, 2020-11-10:
nice, how did you build and publish the package?

heitzmann, 2020-11-10:
@willsALMANJ was the one who created the package through https://github.com/conda-forge/gdspy-feedstock
You can check out how to contribute new recipes to conda forge at https://conda-forge.org/#contribute

joamatab, 2020-11-10:
great, thank you for the links Lucas and @willsALMANJ for making the gdspy conda package

im trying to upload the conda package for a package that I maintain that is based on gdspy
So far I've tried making a github action that gets triggered on each release

https://github.com/gdsfactory/gdsfactory/blob/master/.github/workflows/publish_conda.yml
https://github.com/gdsfactory/gdsfactory/blob/master/conda/meta.yaml

As well as making a PR for the conda forge channel

https://github.com/conda-forge/staged-recipes/pull/13189

What i'm finding is that there are a few dependencies not available in conda (klayout, phidl, pyqtree, picwriter)
https://github.com/conda-forge/staged-recipes/pull/13179

Let me know if you have any ideas / workarounds to install packages that are not yet available in conda

willsALMANJ, 2020-11-10:
Are you interested in packaging those other dependencies for conda yourself? There is a note in the conda-forge documentation about how to submit multiple dependent packages in a single PR: https://conda-forge.org/docs/maintainer/adding_pkgs.html#adding-multiple-packages-at-once

This doesn't help you at all, but for what it's worth I created a flatpak (Linux package) for klayout: https://github.com/flathub/de.klayout.KLayout

joamatab, 2020-11-12:
Hi @willsALMANJ 

I submitted separate PRs for each package, (phidl, pyqtree and picwriter)

for klayout i haven't figured out how to do a conda-forge recipe, let me know if you find a way to do it 

at least I was able to submitt a PR for building the wheel for python3.9, 
https://github.com/KLayout/klayout/pull/672

willsALMANJ, 2020-11-12:
I think it would take some time to get right because KLayout takes quite a while to compile (I think I remember it taking at least 30 minutes on my laptop), so trial and error would take some time, especially if trying to get it working on all three platforms. The build scripts in the KLayout repo lay out how to build it. Ideally, you just have to identify the conda package names for the dependencies (which I think are just Qt, zlib, python, ruby, and the compiler) and have conda-build run the build scripts, but it would probably take some iterating to get right.

closed 2021-01-22 by heitzmann