heitzmann/gdspy #158: Using CellArray gives empty output

SandboChang, 2020-11-23:
Hi,

I am new to this library but so far I have got good experience using it, first of all thanks for the effort.

I have finished my chip level design, so I am moving on to make it into an array, say 3x3, to form a block of chips.
However, when writing the gds file it always writes a 1kB empty file when the array is added to the cell.

More details of the operation:
To do the above, I first write a gds file from my chip.py to give chip.gds.
Then in another python file, block.py, I try to read the chip.gds and assemble the block, here is what I have tried:

```
gdspy.current_library = gdspy.GdsLibrary()

# create block
device_BLK = gdspy.GdsLibrary()

# import chip
os.chdir(r'C:\chipFolder')
device = gdspy.GdsLibrary(infile = 'chip.gds')
device_chip = device.top_level()[0]

# create array of block of 3x3 chips as a cell in device_BLK
device_array = device_BLK.new_cell("Device_Array")
device_array.add(gdspy.CellArray(device_chip , 3, 3, (5000, 0) ) ) # with the array, this does not work
# device_array.add(device_chip) # a test without the array, this works but only write 1 chip to test.gds

# write the block
os.chdir(r'C:\blockFolder')
device_BLK.write_gds('test.gds')
````

Appreciated if any advice can be provided.

heitzmann, 2020-11-23:
@SandboChang I'm glad you find gdspy useful!

You don't need to create a separate library if don't want to (for example, if there are other structures in “chip.gds” that you don't want to copy). So you could just load the existing file into a library, add the array to it, and save as a new file. This makes the code simpler:
```python
import gdspy

# import chip
os.chdir(r'C:\chipFolder')
device_BLK = gdspy.GdsLibrary(infile = 'chip.gds')
device_chip = device_BLK.top_level()[0]

# create array of block of 3x3 chips as a cell in device_BLK
device_array = device_BLK.new_cell("Device_Array")
device_array.add(gdspy.CellArray(device_chip , 3, 3, (5000, 0)))

# write the block
os.chdir(r'C:\blockFolder')
device_BLK.write_gds('test.gds')
```
Otherwise, if you want to filter out other structures by creating an empty library, you must also add your array dependencies to it:
```python
import gdspy

# create block
device_BLK = gdspy.GdsLibrary()

# import chip
os.chdir(r'C:\chipFolder')
device = gdspy.GdsLibrary(infile = 'chip.gds')
device_chip = device.top_level()[0]

# create array of block of 3x3 chips as a cell in device_BLK
device_array = device_BLK.new_cell("Device_Array")
device_array.add(gdspy.CellArray(device_chip , 3, 3, (5000, 0)))

# add dependencies to new library
device_BLK.add(device_array.get_dependencies(True))

# write the block
os.chdir(r'C:\blockFolder')
device_BLK.write_gds('test.gds')
```
You can also avoid manually adding the dependencies by adding the cell after its contents are created:
```python
import gdspy

# import chip
os.chdir(r'C:\chipFolder')
device = gdspy.GdsLibrary(infile = 'chip.gds')
device_chip = device.top_level()[0]

# create array of block of 3x3 chips as a cell in device_BLK
device_array = gdspy.Cell("Device_Array", exclude_from_current=True)
device_array.add(gdspy.CellArray(device_chip , 3, 3, (5000, 0)))

# create block with array
device_BLK = gdspy.GdsLibrary()
device_BLK.add(device_array) # includes dependencies by default

# write the block
os.chdir(r'C:\blockFolder')
device_BLK.write_gds('test.gds')
```

SandboChang, 2020-11-23:
Thanks a lot for your detailed reply (and fast response)!
By following the first set of code you provided, it works like a charm.

I am trying to understand what made it not work in my previous code, was it because I created a separate library?
It seems the main difference was that I separated the two libraries: device and device_BLK.

closed 2020-11-23 by SandboChang

heitzmann, 2020-11-23:
> Thanks a lot for your detailed reply (and fast response)!
> By following the first set of code you provided, it works like a charm.
>
> I am trying to understand what made it not work in my previous [...]

When you create a new library, you have to add the chip structure to it; you were adding only the cell with the array reference, but not the referenced cell.

SandboChang, 2020-11-23:
> When you create a new library, you have to add the chip structure to it; you were adding only the cell with the array reference, but not the referenced cell.

Thanks, I get it now.
Sorry for the many questions, if I may ask one last more, how should I perform the above (say the first set of code) to add two chips onto one block?

I tried with chip1.gds and chip2.gds, and if I repeat this section of code for the two chips:
```
# import chip
os.chdir(r'C:\chipFolder')
device = gdspy.GdsLibrary(infile = 'chip.gds')
device_chip = device.top_level()[0]

# create array of block of 3x3 chips as a cell in device_BLK
device_array = device_BLK.new_cell("Device_Array")
device_array.add(gdspy.CellArray(device_chip , 3, 3, (5000, 0)))
```
It looks like only the chip2 (added later) was kept, and chip1 (added first) was gone when this line was executed:
```
TWPA_BLK.read_gds(infile = 'TWPA_d3b_SG_PN17mod20G18_3WM_5by5.gds')
```.

heitzmann, 2020-11-24:
Technically you should be able to read both files into a single library:

```python
device_BLK = gdspy.GdsLibrary(infile="chip.gds")
device_BLK.read_gds("chip2.gds")

# Since you loaded both files, you'll have (at least) 2 top level cells, presumably from chip and chip2
device_chip = device.top_level()

device_array = device_BLK.new_cell("Device_Array")
# Add first device
device_array.add(gdspy.CellArray(device_chip[0], 3, 3, (5000, 0)))
# Add second device
device_array.add(gdspy.CellArray(device_chip[1], 3, 3, (5000, 0), (15000, 0))
```

There could be a clash of names (each cell must have a unique name in the library). In that case, you could use one of the rename strategies in [`read_gds`](https://gdspy.readthedocs.io/en/stable/reference.html#gdspy.GdsLibrary.read_gds), for example:

```python
device_BLK = gdspy.GdsLibrary(infile="chip.gds", rename_template="{name}-chip1")
device_BLK.read_gds("chip2.gds", rename_template="{name}-chip2")
```

SandboChang, 2020-11-25:
> Technically you should be able to read both files into a single library:
>
> ```python
> device_BLK = gdspy.GdsLibrary(infile="chip.gds")
> device_BLK.read_gds("chip2.gds")
>
> # Since you loaded both files, [...]

Thanks again, the naming was indeed an issue, and by using the rename_template method it works now to have both chip imported. However, I observed a strange (maybe expected?) behaviour:
If I execute the above code from an empty kernel, the device_chip will contain at index 0: chip, index 1: chip2
However if I execute the code again without resetting the kernel, device_chip becomes at index 0: chip2, index 1: chip

In fact, it seems the order is toggled everytime I execute the code. Is it the correct behaviour? 
Would there be a good method then to make sure I create the array with the correct chip? As the index may not be used to determine which chip is which device in this case.

heitzmann, 2020-11-25:
The best way to use the right cell is to get it by name:
```python
device_chip1 = device_BLK.cells["CELL NAME FOR CHIP 1"]
device_array.add(gdspy.CellArray(device_chip1, 3, 3, (5000, 0)))
```

SandboChang, 2020-11-26:
> The best way to use the right cell is to get it by name:
> 
> ```python
> device_chip1 = device_BLK.cells["CELL NAME FOR CHIP 1"]
> device_array.add(gdspy.CellArray(device_chip1, 3, 3, (5000, 0)))
> ```

Thanks again, this is perfect.