heitzmann/gdspy #159: gdspy failed to read 63Gb file and save it to SVG format.

WhitemanE0, 2020-11-25:
I've created a python script as simple as followng:

import numpy
import gdspy
lib = gdspy.GdsLibrary(infile=chip.gds')
cell = lib.cells["topcell"]
cell.write_svg('chip.gds.svg')

With this script I successfully exported .svg for small GDS, but on 63GB

heitzmann, 2020-11-25:
@WhitemanE0 The problem there is not size. From the error message, you have a control character in the middle of what should be an ascii string in your file. It would seem the file is not conformant to the GDSII specification.

WhitemanE0, 2020-11-25:
I can neither confirm it, neither refute.
I just double checked, file is being opened successfully in MGC Calubre DRV without any warnings or errors. 
Do you have any idea, how can I fix this potential problem in my gds to move farther?

heitzmann, 2020-11-25:
Are you generating this file elsewhere? Can you check the string in the layout (cell names and labels) to see if they are all ascii?

You can also change the decode argument to utf8 and see if it works (in the source file gdsiiformat.py).

Or add a try block to it and print the offending string on error.

WhitemanE0, 2020-11-25:
Thank you very much for your reply.
So the problem may be due to Unicode character somewhere in the cell names or layout labels? Right?
Layout database is tsmc28hpc+ ASIC developed with Cadence Innovus, but contains third-party analog blocks like PLL, PCIe, DDR3 etc.

heitzmann, 2020-11-25:
That's my guess, yes. It could also be in some custom property, they too use strings. If you manage to print the string, it should be clear what it is.

WhitemanE0, 2020-12-03:
> Are you generating this file elsewhere? Can you check the string in the layout (cell names and labels) to see if they are all ascii?
> 
> You can also change the decode argument to utf8 and see if it works (in the source file gdsiiformat.py).
> 
> Or add a try block to it and print the offending string on error (so that you know where it is in the file around line 93):
> 
> ```python
>                     if data[-1] == 0:
>                         try:
>                             data = data[:-1].decode("ascii")
>                         except:
>                             print("Offending data:", data[:-1])
>                             data = data[:-1].decode("utf8")
>                     else:
>                         try:
>                             data = data.decode("ascii")
>                         except:
>                             print("Offending data:", data)
>                             data = data.decode("utf8")
> ```
I have edited gdsiiformat.py as you adviced. Now I have the following:

Traceback (most recent call last):
  File "gds2svg.py", line 4, in <module>
    lib = gdspy.GdsLibrary(infile='VM11_2.gds')
  File "/home/v.belyaev/.local/lib/python3.6/site-packages/gdspy/library.py", line 2181, in __init__
    self.read_gds(infile, **kwargs)
  File "/home/v.belyaev/.local/lib/python3.6/site-packages/gdspy/library.py", line 2470, in read_gds
    for record in _record_reader(infile):
  File "/home/v.belyaev/.local/lib/python3.6/site-packages/gdspy/gdsiiformat.py", line 94, in _record_reader
    data = data[:-1].decode("utf8")
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x80 in position 32: invalid start byte

heitzmann, 2020-12-03:
There is definitely something strange with your file, but without more information, I cannot guess what it is. Have you checked on the original program if all strings are ASCII only? Do you have enough memory to load the whole file (just in case it is some weird memory issue)?

closed 2021-01-22 by heitzmann