heitzmann/gdspy #160: gdsPy inversion by Boolean operation results in erroraneous geometry

SandboChang, 2020-12-13:
As I will need to change the process from etching to evaporation, I am trying to invert my design in gdsPy.
To do that, I have a square representing my chip subtracting the original design in an object.path,
```
# original design
main_testL.end()

# attempt to invert it
main_inv = gdspy.boolean(cirAl, main_testL.path, 'not', precision=0.0001, max_points=1999)
cell.add(main_inv)
```

However, the resulted gds file looks strange. From the same original file, however I could get the expected output from KLayout.
Here is a comparison between original vs gdsPy vs KLayout.
https://imgur.com/undefined

heitzmann, 2020-12-13:
Yes, you can use only the polygons in a specific layer. Just build a list of all the polygons in the layer you want. Alternatively, you can use [`gdspy.Cell.get_polygons(by_layer=True)`](https://gdspy.readthedocs.io/en/stable/reference.html#gdspy.Cell.get_polygons) if you'll be using all polygons in a specific cell.

If you can share more information, maybe I can help look into the issue.

SandboChang, 2020-12-13:
> Yes, you can use only the polygons in a specific layer. Just build a list of all the polygons in the layer you want. Alternatively, you can use [...]

Thanks for getting back to me. I wish I can provide the code but it is rather long and messy at the moment.
As I have a deadline tomorrow, for now I would use KLayout to perform inversion, then I will return to the issue and study it in more depth.

SandboChang, 2020-12-15:
Allow me to close it for now, when I have an update I will try to reopen it.

closed 2020-12-15 by SandboChang