heitzmann/gdspy #162: gdspy filed to remove references on deleted lib.cells

WhitemanE0, 2020-12-19:
I wrote simple script:
import numpy
import gdspy
import sys

filename=''

del sys.argv[0]

try:
	filename=str(sys.argv[0])
except:
	print('Must provide a filename')
	exit()

for file in sys.argv:
	filename=str(file)
	print('Processing '+filename)
	lib = gdspy.GdsLibrary(infile=filename)
	print(lib)
	cell = lib.cells['ASIC_pad_ring']
	print(lib.cells)
	cellsToRemove = []
	for c in lib.cells:
		if c == 'ASIC_pad_ring':
			print('match!!')	
		else:	
			cellsToRemove.append(c)
	print('cells to remove: ',cellsToRemove)
	for r in cellsToRemove:
		lib.remove(r, remove_references=True)

	print('Library: ',lib)
	print('Remaining Library cells: ',lib.cells)
	print('Remaining cell: ',cell)

##########################
##########################
Output from "python3 gds2svg.py io_ring_16ffc.gds"

Processing io_ring_16ffc.gds
/home/e0/.local/lib/python3.8/site-packages/gdspy/library.py:2181: UserWarning: [GDSPY] GDSII elements of type BOX are imported as polygons.
  self.read_gds(infile, **kwargs)
GdsLibrary (ASIC_pad_ring, ASIC_pad_ring_VIA0, ASIC_pad_ring_VIA1, ASIC_pad_ring_VIA2, ASIC_pad_ring_VIA3, ASIC_pad_ring_VIA4, ASIC_pad_ring_VIA5, ASIC_pad_ring_VIA6, HBR177_V11, HBR177_VIAZ_TOP, HBR177_VIAA, HBR177_VIAZ_INTER, HBR177_V8, HBR177_V7, HBR177_VIAYY, HBR177_V9, HBR177_V6, HBR177_VIAXY, HBR177_VIAY, HBR177_V5, HBR177_VIAC, HBR177_VIAC_1, HBR177_V10, HBR177_VIAZ_FIRST, HBR177_VIAB, HBR177_V4, HBR177_V4X11, HBR177_V4X7, HBR177_VIA_PWRMID_IO, HBR177_VIA_PWRMID_PG ...etc}
match!!
cells to remove:  ['ASIC_pad_ring_VIA0', 'ASIC_pad_ring_VIA1', 'ASIC_pad_ring_VIA2', 'ASIC_pad_ring_VIA3', 'ASIC_pad_ring_VIA4', 'ASIC_pad_ring_VIA5', 'ASIC_pad_ring_VIA6', 'HBR177_V11', 'HBR177_VIAZ_TOP', 'HBR177_VIAA', 'HBR177_VIAZ_INTER', 'HBR177_V8', 'HBR177_V7', 'HBR177_VIAYY', 'HBR177_V9', 'HBR177_V6', 'HBR177_VIAXY', 'HBR177_VIAY', 'HBR177_V5', 'HBR177_VIAC', 'HBR177_VIAC_1', 'HBR177_V10', 'HBR177_VIAZ_FIRST' ...etc]
Library:  GdsLibrary (ASIC_pad_ring)
Remaining Library cells:  {'ASIC_pad_ring': <gdspy.library.Cell object at 0x7fde72bed9a0>}
Remaining cell:  Cell ("ASIC_pad_ring", 189 polygons, 0 paths, 224 labels, 156 references)

Output shows, that all cells exvept one was removed, but in cell 'ASIC_pad_ring' there are 156 references left, despite "remove_references=True" argument in lib.remove(r, remove_references=True)

WhitemanE0, 2020-12-19:
I have workarouded it by adding 
`lib.cells['ASIC_pad_ring'].references = []`

But please explain if the initial reported is a bug or expected behavior

heitzmann, 2020-12-19:
The expected behavior is to exclude all references to removed cell. These remaining references are not being found when the other cells are removed, but I'm not sure why.

Could you run this version with some extra debug information for us to see what is happening? (Or, if you can send me the gds file, I can take a look at its internals to figure out the problem.)

```python
import gdspy
import sys

if len(sys.argv) < 2:
    print("Must provide a filename")
    exit(1)

cellToKeep = "ASIC_pad_ring"

for filename in sys.argv[1:]:
    print(f"Processing {filename}")

    lib = gdspy.GdsLibrary(infile=filename)
    print("Original library:", lib)
    print("Original library cells:", lib.cells)

    if cellToKeep not in lib.cells:
        print(f"{cellToKeep} not found!")
        continue

    cellsToRemove = list(lib.cells.keys())
    cellsToRemove.remove(cellToKeep)
    print("Cells to remove:", cellsToRemove)

    for ref in lib.cells[cellToKeep].references:
        refCellName = (
            ref.ref_cell.name if isinstance(ref.ref_cell, gdsp.Cell) else ref.ref_cell
        )
        if refCellName not in cellsToRemove:
            print(
                f"Reference {ref} will NOT be removed: {refCellName} not in cellsToRemove."
            )

    for r in cellsToRemove:
        lib.remove(r, remove_references=True)

    print("Library:", lib)
    print("Remaining library cells:", lib.cells)
    print("Remaining cell:", lib.cells[cellToKeep])
```

Besides that, if all you want is to make new library with a single cell and no references, this is much more efficient:

```python
import gdspy
import sys

if len(sys.argv) < 2:
    print("Must provide a filename")
    exit(1)

cellToKeep = "ASIC_pad_ring"

for filename in sys.argv[1:]:
    print(f"Processing {filename}")

    lib = gdspy.GdsLibrary(infile=filename)
    print("Original library:", lib)
    print("Original library cells:", lib.cells)

    if cellToKeep not in lib.cells:
        print(f"{cellToKeep} not found!")
        continue

    cell = lib.cells[cellToKeep]
    cell.references = []
    newLib = gdspy.GdsLibrary()
    newLib.add(cell)

    print("Library:", newLib)
    print("Remaining library cells:", newLib.cells)
    print("Remaining cell:", newLib.cells[cellToKeep])
```

WhitemanE0, 2020-12-19:
Seems you've fixed my code, this is output of first part:
Processing io_ring_16ffc.gds
Original library: GdsLibrary (ASIC_pad_ring, ASIC_pad_ring_VIA0, ASIC_pad_ring_VIA1, ASIC_pad_ring_VIA2, ASIC_pad_ring_VIA3, PAD180PITCH, HBR177_V11, HBR177_VIAZ_TOP, HBR177_VIAA, HBR177_VIAZ_INTER, HBR177_V8, HBR177_V7, HBR177_VIAYY, HBR177_V9, HBR177_V6, HBR177_VIAXY, HBR177_VIAY, HBR177_V5, HBR177_VIAC, HBR177_VIAC_1, HBR177_V10, HBR177_VIAZ_FIRST, HBR177_VIAB, HBR177_V4, HBR177_V4X11, HBR177_V4X7, HBR177_VIA_PWRMID_IO, HBR177_VIA_PWRMID_PG, HBR177_BASE_CORNER_VIA, PCORNERSD, HBR177_VT_PWR, HBR177_BASE_PWR_IO_MC_1, HBR177_BASE_IO_PWR_MC_1, HBR177_BASE_PWR_IO_MC, HBR177_BASE_PWR_IO_SIGNAL, HBR177_PWR_IO_MC, HBR177_BASE_PWR_IO_MB, HBR177_PWR_IO_MB, HBR177_BASE_PWR_IO_MA, HBR177_PWR_IO_MA, HBR177_VIAPAD, HBR177_PAD_IO, HBR177_PWR_IO, HBR177_V1S_BASE, HBR177_V1S, HBR177_V2S_BASE, HBR177_V2S, HBR177_V12S, HBR177_V3S_BASE, HBR177_V3S, HBR177_V23S, HBR177_V0, HBR177_V0S, HBR177_V0_MP04, HBR177_FIN_OD_D118, HBR177_MOS_WD538_LD180_NOV0, HBR177_MOS_IOH_POST_CAS1, HBR177_GR_WD538_LD086_NOMP, HBR177_MP_D06_D306, HBR177_GR_WD538_LD086_NOV0, HBR177_MOS_IOH_POST_CAS2, HBR177_MOS_WD538_LD180_PODE_0, HBR177_UNIT_IOH_POST_CAS, HBR177_BUS_IOH_POST_CAS_HF, HBR177_BUS_IOH_POST_CAS, HBR177_PAIR_IOH_POST_CAS_WIBUS, HBR177_GR_WD202_LD086_NOMP, HBR177_GR_WD202_LD086_NOV0, HBR177_GR_WD202_LD180_NOMP, HBR177_GR_WD202_LD180_NOV0, HBR177_SUB_IOH_POST_CAS, HBR177_SUB_IOH_POST_CASN, HBR177_BUS_IOH_POST_PAD_MC, HBR177_V123S, HBR177_BUS_IOH_POST_PAD_HF, HBR177_BUS_IOH_POST_PAD, HBR177_BUS_IOH_POST_DGR, HBR177_MOS_WD250_LD135_PODE, HBR177_MOS_WD922_LD150_NOV0, HBR177_MOS_WD922_LD150_NOVG, HBR177_MOS_WD922_LD150, HBR177_MOS_WD922_LD150_PODE, HBR177_DODPO_WD202_LD086_RES, HBR177_RES_W1D642_L1D642_BASE, HBR177_RES_W1D642_L1D642, HBR177_DODPO_WD202_LD086, HBR177_GR_WD202_LD150_NOMP, HBR177_GR_WD202_LD150_NOV0, HBR177_GR_WD922_LD086_NOMP, HBR177_GR_WD922_LD086_NOV0, HBR177_V3_BASE, HBR177_MOS_WD250_LD240_NOV0, HBR177_MOS_WD250_LD240_NOVG, HBR177_SUB_IOH_POST_TRACK, HBR177_SUB_IOH_POST_CASP, HBR177_DIO_HIA_WD346_BASE, HBR177_DIO_HIA_WD346_GR, HBR177_DIO_HIA_WD346_HF, HBR177_DIO_HIA_WD346, HBR177_BASE_IOH_POST_HIA, HBR177_UNIT_IOH_POST_NHIA, HBR177_V3, HBR177_MASH_IOH_PAD, HBR177_UNIT_IOH_POST_PHIA, HBR177_IOH_POSTDRV, HBR177_PAD_JOG_M567890, HBR177_BASE_IOH_GR_STRAP, HBR177_GR_IOH_HF_M234, HBR177_GR_IOH_HF, HBR177_GR_IOH, HBR177_V2_BASE, HBR177_V2, HBR177_V1_BASE, HBR177_V1, HBR177_V123, HBR177_PIN_D26_M1234, HBR177_FIN_TEXT, HBR177_V2_4X4, HBR177_V2_4X2, HBR177_BASE_IOVH_PREDRV_PW, HBR177_BUS_IOVH_PREDRV_MSC33, HBR177_BUS_IOVH_PREDRV_18L, HBR177_SUB_IOVH_PREDRV_PW, HBR177_BASE_IOVH_PREDRV_HIA, HBR177_BASE_IOH_PREDRV_HIA, HBR177_UNIT_IOH_PREDRV_PHIA, HBR177_UNIT_IOH_PREDRV_NHIA, HBR177_V01SM1, HBR177_RES_L1D888_WD5_BASE, HBR177_RES_IOH_PAD2_30K, HBR177_MOS_WD202_LD180_NOV0, HBR177_MOS_WD202_LD180_NOVG, HBR177_MOS_WD202_LD180, HBR177_MOS_WD922_LD135_PODE, HBR177_MOS_WD250_LD150_PODE, HBR177_BUS_IOVH_PREDRV_MSC33_M1, HBR177_GR_WD826_LD086_NOMP, HBR177_GR_WD826_LD086_NOV0, HBR177_MOS_WD826_LD150_PODE, HBR177_MOS_WD826_LD150_NOV0, HBR177_MOS_WD826_LD150_NOVG, HBR177_MOS_IOV_PREDRV_MSC33, HBR177_MOS_IOV_PREDRV_18L, HBR177_GR_WD202_LD086, HBR177_SUB_IOV_PREDRV_18L, HBR177_V1SX3, HBR177_V1SX2, HBR177_MOS_WD922_LD240_NOV0, HBR177_MOS_WD922_LD240_NOVG, HBR177_MOS_WD922_LD240, HBR177_BASE_IOV_INPUT_DELAY_CAS, HBR177_MOS_WD202_LD150_NOV0, HBR177_MOS_WD202_LD150_NOVG, HBR177_MOS_WD202_LD150, HBR177_MOS_WD202_LD150_PODE, HBR177_BASE_IOV_INPUT_DELAY, HBR177_SUB_IOV_INPUT_DELAY, HBR177_MOS_WD250_LD150_NOV0, HBR177_MOS_WD250_LD150_NOVG, HBR177_MOS_WD250_LD150, HBR177_MOS_WD202_LD135_PODE, HBR177_MOS_WD154_LD135_PODE, HBR177_MOS_WD442_LD150_PODE, HBR177_MOS_WD154_LD150_NOV0, HBR177_MOS_WD154_LD150_NOVG, HBR177_BASE_IOV_LATCH_RTE, HBR177_MOS_WD154_LD150_PODE, HBR177_MOS_WD154_LD150, HBR177_BASE_IOV_LATCH_HF, HBR177_BASE_IOV_LATCH, HBR177_SUB_IOV_IP_LATCH, HBR177_MOS_WD442_LD150_NOV0, HBR177_MOS_WD442_LD150_NOVG, HBR177_MOS_WD442_LD150, HBR177_MOS_WD682_LD150_PODE, HBR177_BASE_IOV_LATCH_HF_POC0, HBR177_BASE_IOV_LATCH_POC0_1, HBR177_MOS_WD682_LD150_NOV0, HBR177_MOS_WD682_LD150_NOVG, HBR177_MOS_WD682_LD150, HBR177_SUB_IOV_PU_LATCH_POC0, HBR177_SUB_IOV_IE_LATCH_POC0, HBR177_MOS_WD538_LD150_PODE, HBR177_MOS_WD538_LD150_NOV0, HBR177_MOS_WD538_LD150_NOVG, HBR177_MOS_WD538_LD150, HBR177_MOS_WD346_LD150_PODE, HBR177_MOS_WD346_LD150_NOV0, HBR177_MOS_WD346_LD150_NOVG, HBR177_SUB_IOV_SMIT_PSG, HBR177_MOS_WD682_LD135_PODE, HBR177_MOS_WD682_LD180_NOV0, HBR177_MOS_WD682_LD180_NOVG, HBR177_SUB_IOV_PAD2C_IN, HBR177_GR_WD490_LD086_NOMP, HBR177_GR_WD490_LD086_NOV0, HBR177_MOS_WD298_LD150_PODE, HBR177_MOS_WD298_LD150_NOV0, HBR177_MOS_WD298_LD150_NOVG, HBR177_MOS_WD298_LD150, HBR177_MOS_WD586_LD150_PODE, HBR177_MOS_WD586_LD150_NOV0, HBR177_MOS_WD586_LD150_NOVG, HBR177_MOS_WD586_LD150, HBR177_SUB_IOV_SMIT_TIGGER, HBR177_SUB_IOV_SMIT_PSG_1, HBR177_MOS_WD538_LD135_PODE, HBR177_SUB_IOV_PAD2C, HBR177_SUB_IOV_CONTROL_RTEPOCMS, HBR177_SUB_IOV_DS_NGATE, HBR177_SUB_IOV_ST_LATCH_POC0, HBR177_BASE_IOV_LATCHN, HBR177_SUB_IOV_I_LATCH, HBR177_BASE_IOV_LATCH_POC0, HBR177_SUB_IOV_PD_LATCH_POC0, HBR177_MOS_WD346_LD150, HBR177_SUB_IOV_DS_LATCH_POC0, HBR177_SUB_IOV_PREDRV_NG_POC0, HBR177_SUB_IOV_PREDRV_NG_WIM2_POC0, HBR177_DCPO_WD922_LD024, HBR177_DODPO_WD922_LD024, HBR177_CPO_WD922_LD024, HBR177_MOS_WD922_LD024_PODE, HBR177_FIN_OD_D072, HBR177_MOS_WD922_LD024_NOV0, HBR177_MOS_WD922_LD024_NOVG, HBR177_BAS [... cut ...] _LD36_EDGE', 'HBR177_RES_TIN_W8D026_LD36_BASE', 'HBR177_RES_TIN_W8D026_LD36XN', 'HBR177_SUB_LDO_CONN', 'HBR177_BASE_LDO', 'HBR177_SUB_LDOV', 'HBR177_PVDD3LDOM_V_TEXT_CELL', 'HBR177_PVDD3LDOM_V_VOLTAGE_CELL', 'HBR177_VT_LDOV', 'PVDD3LDOM_V', 'lvds_ref_pad', 'lvds_tx_pad', 'lvds_rx_pad']
Library: GdsLibrary (ASIC_pad_ring)
Remaining library cells: {'ASIC_pad_ring': <gdspy.library.Cell object at 0x7f59435f99a0>}
Remaining cell: Cell ("ASIC_pad_ring", 172 polygons, 0 paths, 224 labels, 0 references)

WhitemanE0, 2020-12-19:
And thank you for the master-class.
I'm only doing my first steps with python :)

heitzmann, 2020-12-19:
No problem! Glad it works now!

closed 2020-12-19 by heitzmann