heitzmann/gdspy #163: Is there any way to show progress on .flatten()

WhitemanE0, 2020-12-19:
I'm running .flatten() on a cell and it lasts very long	

I do like:

```
topCell = lib.top_level()[0]
topCell = topCellList[0].flatten()

```
Is there any way to report progress while running .flatte() ?

heitzmann, 2020-12-19:
@WhitemanE0 You seem to be overwriting `topCell` for no reason... I'm guessing your first line should read `topCellList = lib.top_level()`.

Anyway, flatten does take a long time if your layout is large. It has to create all polygon copies from a multiple hierarchy of references, so there is little we can accelerate. Printing a progress report would only slow down the process even more, but you can do it by putting some `print` calls directly into the source code.

There are a few functions that are computationally expensive: boolean, inside, offset, flatten and (to some extent) bounding box calculations are the ones I remember from the top of my head. It is usually better to use them only as a last resort.

WhitemanE0, 2020-12-19:
topCellList = lib.top_level() return a list with 1 element in my case

WhitemanE0, 2020-12-19:
It will slowdown, but progress will be definately useful
`27410 v.belya+  20   0   81,3g  79,8g   8436 R 100,0 15,8  84:10.95 python3 `
it is running an hour and a half, took 80g RAM, and there is no idea for how long it will work on the file 1 hour or 10 hours.

I have updated my library.py as following:
```
    def flatten(self, single_layer=None, single_datatype=None, single_texttype=None):
        """
        Convert all references into polygons, paths and labels.

        Parameters
        ----------
        single_layer : integer or None
            If not None, all polygons will be transfered to the
            layer indicated by this number.
        single_datatype : integer or None
            If not None, all polygons will be transfered to the
            datatype indicated by this number.
        single_datatype : integer or None
            If not None, all labels will be transfered to the
            texttype indicated by this number.

        Returns
        -------
        out : `Cell`
            This cell.
        """
        self.labels = self.get_labels()
        if single_layer is not None and single_datatype is not None:
            i = 1
            for lbl in self.labels:
                print('Working on '+lbl+' which is the '+i+' of '+len(self.labels))
                lbl.layer = single_layer
                lbl.texttype = single_texttype
                i = i + 1
        elif single_layer is not None:
            i = 1
            for lbl in self.labels:
                print('Working on '+lbl+' which is the '+i+' of '+len(self.labels))
                lbl.layer = single_layer
                i = i + 1
        elif single_datatype is not None:
            i = 1
            for lbl in self.labels:
                print('Working on '+lbl+' which is the '+i+' of '+len(self.labels))
                lbl.texttype = single_texttype
                i = i + 1
        self.polygons = self.get_polygonsets()
        self.paths = self.get_paths()
        if single_layer is not None and single_datatype is not None:
            i = 1
            for poly in self.polygons:
                print('Working on '+poly+' which is the '+i+' of '+len(self.polygons))
                poly.layers = [single_layer] * len(poly.polygons)
                poly.datatypes = [single_datatype] * len(poly.polygons)
                i = i + 1
            i = 1
            for path in self.paths:
                print('Working on '+path+' which is the '+i+' of '+len(self.paths))
                path.layers = [single_layer] * path.n
                path.datatypes = [single_datatype] * path.n
                i = i + 1
        elif single_layer is not None:
            i = 1
            for poly in self.polygons:
                print('Working on '+poly+' which is the '+i+' of '+len(self.polygons))
                poly.layers = [single_layer] * len(poly.polygons)
                i= i + 1
            i = 1
            for path in self.paths:
                print('Working on '+path+' which is the '+i+' of '+len(self.paths))
                path.layers = [single_layer] * path.n
                i = i + 1
        elif single_datatype is not None:
            i = 1
            for poly in self.polygons:
                print('Working on '+poly+' which is the '+i+' of '+len(self.polygons))
                poly.datatypes = [single_datatype] * len(poly.polygons)
            i = 1
            for path in self.paths:
                print('Working on '+path+' which is the '+i+' of '+len(self.paths))
                path.datatypes = [single_datatype] * path.n
                i = i + 1
        self.references = []
        return self
```
Have not tried to run yet, maybe there is a typos, but don't want to cancel 1:30 run in progress.

heitzmann, 2020-12-19:
The progress indication in this function is not going to help you much. The bottleneck is `self.polygons = self.get_polygonsets()`. Then, iteratively going through all references' `get_polygons()`… The progress indicators will have to be placed in those functions as well.

closed 2021-01-22 by heitzmann