heitzmann/gdspy #165: iteration through the cell.labels, cell.polygons, cell.path

WhitemanE0, 2020-12-28:
I did the following script to remove unneeded layers from GDS:

`import numpy
import gdspy
import sys
import copy

layersToRemove = [ 170, 16, 3, 6, 36, 37, 38, 39, 59, 63, 85, 108, 111, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140]

if len(sys.argv) < 2:
	print("Must provide a filename")
	exit(1)

for filename in sys.argv[1:]:
	print(f"Processing {filename}")

	lib = gdspy.GdsLibrary(infile=filename)

	top = lib.cells['ASIC_pad_ring']
	print("Original library:", top)

	i = 0
	for l in top.labels:
		if l.layer in layersToRemove:
			del top.labels[i]
		i += 1
	i = 0
	for l in top.polygons:
		if l.layers[0] in layersToRemove:
			del top.polygons[i]
		i += 1
	i = 0
	for l in top.paths:
		if l.layers[0] in layersToRemove:
			
			del top.paths[i]
		i += 1
	print("Modifyed Library", top)
	lib.write_gds('final.gds')
	top.write_svg('ASIC_pad_ring.svg')
`

But I still see the unneded layers in the GDS.

When I re-run the script on GDS recieved, I see the following:

`Processing ASIC_pad_ring_flatten.gds
Original library: Cell ("ASIC_pad_ring", 88212 polygons, 131877 paths, 245824 labels, 0 references)
Modifyed Library Cell ("ASIC_pad_ring", 56700 polygons, 85269 paths, 122989 labels, 0 references)
[v.belyaev@proto0 tsmc16ffc_ioring]$ python3 minify_gds.py final.gds 
Processing final.gds
Original library: Cell ("ASIC_pad_ring", 56700 polygons, 85269 paths, 122989 labels, 0 references)
Modifyed Library Cell ("ASIC_pad_ring", 41127 polygons, 61550 paths, 61571 labels, 0 references)
[v.belyaev@proto0 tsmc16ffc_ioring]$ python3 minify_gds.py final.gds 
Processing final.gds
Original library: Cell ("ASIC_pad_ring", 41127 polygons, 61550 paths, 61571 labels, 0 references)
Modifyed Library Cell ("ASIC_pad_ring", 34368 polygons, 49898 paths, 30862 labels, 0 references)
[v.belyaev@proto0 tsmc16ffc_ioring]$ python3 minify_gds.py final.gds 
Processing final.gds
Original library: Cell ("ASIC_pad_ring", 34368 polygons, 49898 paths, 30862 labels, 0 references)
Modifyed Library Cell ("ASIC_pad_ring", 31164 polygons, 44487 paths, 15508 labels, 0 references)
[v.belyaev@proto0 tsmc16ffc_ioring]$ python3 minify_gds.py final.gds 
Processing final.gds
Original library: Cell ("ASIC_pad_ring", 31164 polygons, 44487 paths, 15508 labels, 0 references)
Modifyed Library Cell ("ASIC_pad_ring", 28996 polygons, 41574 paths, 7831 labels, 0 references)
[v.belyaev@proto0 tsmc16ffc_ioring]$ python3 minify_gds.py final.gds 
Processing final.gds
Original library: Cell ("ASIC_pad_ring", 28996 polygons, 41574 paths, 7831 labels, 0 references)
Modifyed Library Cell ("ASIC_pad_ring", 28273 polygons, 39910 paths, 3992 labels, 0 references)
[v.belyaev@proto0 tsmc16ffc_ioring]$ python3 minify_gds.py final.gds 
Processing final.gds
Original library: Cell ("ASIC_pad_ring", 28273 polygons, 39910 paths, 3992 labels, 0 references)
Modifyed Library Cell ("ASIC_pad_ring", 28120 polygons, 39076 paths, 2072 labels, 0 references)
[v.belyaev@proto0 tsmc16ffc_ioring]$ python3 minify_gds.py final.gds 
Processing final.gds
Original library: Cell ("ASIC_pad_ring", 28120 polygons, 39076 paths, 2072 labels, 0 references)
Modifyed Library Cell ("ASIC_pad_ring", 28078 polygons, 39076 paths, 1113 labels, 0 references)
[v.belyaev@proto0 tsmc16ffc_ioring]$ python3 minify_gds.py final.gds 
Processing final.gds
Original library: Cell ("ASIC_pad_ring", 28078 polygons, 39076 paths, 1113 labels, 0 references)
Modifyed Library Cell ("ASIC_pad_ring", 28077 polygons, 39076 paths, 633 labels, 0 references)
[v.belyaev@proto0 tsmc16ffc_ioring]$ python3 minify_gds.py final.gds 
Processing final.gds
Original library: Cell ("ASIC_pad_ring", 28077 polygons, 39076 paths, 633 labels, 0 references)
Modifyed Library Cell ("ASIC_pad_ring", 28077 polygons, 39076 paths, 393 labels, 0 references)
[v.belyaev@proto0 tsmc16ffc_ioring]$ python3 minify_gds.py final.gds 
Processing final.gds
Original library: Cell ("ASIC_pad_ring", 28077 polygons, 39076 paths, 393 labels, 0 references)
Modifyed Library Cell ("ASIC_pad_ring", 28077 polygons, 39076 paths, 273 labels, 0 references)
[v.belyaev@proto0 tsmc16ffc_ioring]$ python3 minify_gds.py final.gds 
Processing final.gds
Original library: Cell ("ASIC_pad_ring", 28077 polygons, 39076 paths, 273 labels, 0 references)
Modifyed Library Cell ("ASIC_pad_ring", 28077 polygons, 39076 paths, 213 labels, 0 references)
[v.belyaev@proto0 tsmc16ffc_ioring]$ python3 minify_gds.py final.gds 
Processing final.gds
Original library: Cell ("ASIC_pad_ring", 28077 polygons, 39076 paths, 213 labels, 0 references)
Modifyed Library Cell ("ASIC_pad_ring", 28077 polygons, 39076 paths, 183 labels, 0 references)
`

Maybe my code is lame, but plese kindly explain what is the problem.

Smarter way to remove layers from the given list is also higly appreciated.

heitzmann, 2020-12-28:
There are a couple of issues there. First, since you are only checking whether the layer is in `layersToRemove`, it is a lot more efficient to use a `set` than a `list`:

```python
layersToRemove = {170, 16, 3, …, 140}
```

Second, you should not loop through the lists and modify them simultaneously. This would be a more readable if you want to loop manually:

```python
i = 0
while i < len(top.polygons):
    if top.polygons[i].layer in layersToRemove:
        del top.polygons[i]
    else:
        i += 1
```

Finally, these loops will not remove elements in references. If you want those removed as well, you'll have to loop through all dependencies' elements as well. It is also more efficient to use the remove functions:

```python
import gdspy
import sys

if len(sys.argv) < 2:
    print("Must provide a filename")
    exit(1)

layersToRemove = {170, 16, 3, 6, 36, 37, 38, 39, 59, 63, 85, 108, 111, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140}

def cleanup(cell):
    cell.remove_polygons(lambda points, layer, datatype: layer in layersToRemove)
    cell.remove_paths(lambda path: path.layers[0] in layersToRemove)
    cell.remove_labels(lambda label: label.layer in layersToRemove)

for filename in sys.argv[1:]:
    print(f"Processing {filename}")

    lib = gdspy.GdsLibrary(infile=filename)

    top = lib.cells['ASIC_pad_ring']
    print("Original library:", top)

    cleanup(top)
    for dep in top.get_dependencies(True):
        cleanup(dep)

    print("Modifyed Library", top)
    lib.write_gds('final.gds')
    top.write_svg('ASIC_pad_ring.svg')
```

WhitemanE0, 2020-12-28:
Thank you very much. It really does the thing.
And it's a great master class as well!

closed 2020-12-28 by WhitemanE0