heitzmann/gdspy #166: Merge all polygons with same gds number and datatype

WhitemanE0, 2021-01-04:
I hope I have not bothered you to much yet, but I did the script to merge all polygons with same gds number and datatype. This is a useful task, as if there are many intersecting polygons in GDS it can reduce the size of the file and give a cleaner look.
```
import numpy as np
import gdspy
import sys
import copy

filename=''

try:
	filename=str(sys.argv[1])
except:
	print('Must provide a filename')
	print('Usage: $python3 merge_aLl_shapes.py filename.gds')
	exit()
print('Merging all shapes in '+filename)

lib = gdspy.GdsLibrary(infile=filename)

newLib = copy.copy(lib)
newLib.cells = {}

topCell = lib.top_level()[0]#.flatten()

newCellNameSuffix = '_merged_layers'
newLibCell = gdspy.Cell(topCell.name+newCellNameSuffix)

polygons = topCell.polygons

layershashDt={}
polygonSet = {}
roundedPolygons = []

for p in polygons:
	polygonSet[str(p.layers[0])+'_'+str(p.datatypes[0])] = []

for p in polygons:
	roundedPolygons = p.polygons
#	roundedPolygons = np.round(p.polygons, decimals=1)
#	roundedPolygons[0] = np.array(roundedPolygons[0])
	polygonSet[str(p.layers[0])+'_'+str(p.datatypes[0])].extend(roundedPolygons)

for p in polygonSet.keys():
	polygonSetToMerge = gdspy.PolygonSet(polygonSet[p], layer=int(p.split('_')[0]), datatype=int(p.split('_')[1]))
	print(polygonSetToMerge)
	result = gdspy.boolean(polygonSetToMerge, None, "or", layer=int(p.split('_')[0]), datatype=int(p.split('_')[1]))
	try:
		newLibCell.add(result)
	except:
		pass

newLib.add(newLibCell)

print('Initial cell statistics: ',lib.cells[topCell.name])
print('Final cell statistics: ',newLibCell, newLib)

newLib.write_gds(newLibCell.name+'.gds')
#newLibCell.write_svg(newLibCell.name+'.gds.svg')
```
It does the thing, but I will appreciate yet another master-class and suggestion how to make it better and smarter.
P.S. I've forgot to add a text labels to the result, but it easy to fix.

I like yout gdspy module very much and dream to create a command-line front end for most common layout tasks when I'll advance in my development.

heitzmann, 2021-01-04:
@WhitemanE0 I'm glad to hear Gdspy is useful for you!

As for your script, copying the whole library just to clear it afterwards seems to be a waste.
you can eliminate the first loop, that just create empty lists, by using [`collections.defaultdict`](https://docs.python.org/3/library/collections.html#collections.defaultdict) (or a simple if statement to check for the existence of the desired key). Furthermore, it is more efficient to use a tuple `(layer, datatype)` as key, instead of converting to a string (and later converting it back).

Note that you're not operating on other cells included as references (if that's your intent) or on paths.

```python
import numpy as np
import gdspy
import sys
import collections

# If statements are more efficient than try/catch blocks.
# If you use catch, specify which error you're catching, or you could
# end up catching (and ignoring) unexpected errors.
if len(sys.argv) < 2:
    print('Must provide a filename')
    print('Usage: $python3 merge_aLl_shapes.py filename.gds')
    exit()

filename = sys.argv[1]  # sys.argv is already a list of strings

print('Merging all shapes in ' + filename)

lib = gdspy.GdsLibrary(infile=filename)

newLib = gdspy.GdsLibrary(lib.name, unit=lib.unit, precision=lib.precision)

topCell = lib.top_level()[0]

newCellNameSuffix = '_merged_layers'
newLibCell = gdspy.Cell(topCell.name + newCellNameSuffix)

polygons = topCell.polygons
polygonSet = collections.defaultdict(list)

for p in polygons:
	polygonSet[(p.layers[0], p.datatypes[0])].append(p.polygons[0])

for (l, d), poly in polygonSet.items():
	print(f"Layer {l}, datatype {d}: {len(poly)} polygons to merge")
	result = gdspy.boolean(poly, None, "or", layer=l, datatype=d)
	if result is not None:
		newLibCell.add(result)

# This will include all dependencies as well, but those dependencies
# did not go trough the same merging process.
newLib.add(newLibCell)

print('Initial cell statistics:', topCell)
print('Final cell statistics:', newLibCell, newLib)

newLib.write_gds(newLibCell.name + '.gds')
#newLibCell.write_svg(newLibCell.name + '.gds.svg')
```

This version has other clean-ups and process all cells in the library:

```python
import sys
import collections
import gdspy

def merge_layers_and_datatypes(cell):
    print(f"Processing cell {cell.name}")

    polygonSet = collections.defaultdict(list)
    for p in cell.polygons:
        for l, d, points in zip(p.layers, p.datatypes, p.polygons):
            polygonSet[(l, d)].append(points)

    cell.polygons = []

    for (l, d), plist in polygonSet.items():
        if len(plist) > 1:
            print(f"  Layer {l}, datatype {d}: {len(plist)} polygons to merge")
            result = gdspy.boolean(plist, None, "or", layer=l, datatype=d)
        elif len(plist) == 1:
            cell.add(gdspy.Polygon(plist[0], layer=l, datatype=d))

if len(sys.argv) < 2:
    print(f"Missing filename\nUsage: python3 {sys.argv[0]} FILENAME")
    exit(1)

filename = sys.argv[1]
print(f"Merging all shapes in {filename}")

lib = gdspy.GdsLibrary(infile=filename)
for cell in lib.cells.values():
    merge_layers_and_datatypes(cell)
lib.write_gds(f"{filename}_merged.gds")
```

WhitemanE0, 2021-01-04:
Thank you very much for your master-class.
I want to explain why I'm copy entire library and than clean all cells.
If I create newLib from scratch I can not export new gds to fike becouse library does not have 'precision' and other stuff. Probably I can add it manually to newly created lib, but it requires research, this way I have all desired properties in a new library but no objects.
Regarding paths, I ommited it for now  becouse target gds's have only polygons.
Main target was reduce .svg file size, and for my run I used .flatten() on topCell hence I have not lost any references.

Thank you very much for your attantion, I will continue study python and gdspy, it is really fascinating.

closed 2021-01-22 by heitzmann