heitzmann/gdspy #167: how to get svg file of each layer of the cell from gds file

mo-nu, 2021-01-22:
Hi Heitzmann,

Thank you for this great module.

I have a gds file for the library. I am able to obtain the svg files of all cells.  But, is there a way to get a svg file for each layer of the cells separately? 

[svg.zip](https://github.com/heitzmann/gdspy/files/5853348/svg.zip)

Thanks

heitzmann, 2021-01-22:
This should be a good start:

```python
import sys
import copy
import gdspy

if len(sys.argv) < 2:
    sys.stderr.write(f"Missing file name!\nUSAGE: {sys.argv[0]} filename\n")
    sys.exit(1)

for fname in sys.argv[1:]:
    print(f"Processing {fname} layers:", end="", flush=True)

    lib = gdspy.GdsLibrary(infile=sys.argv[1])

    all_layers = set()
    for cell in lib.top_level():
        all_layers.update(cell.get_layers())

    for keep in all_layers:
        print(f" {keep}", end="", flush=True)
        lib_copy = copy.deepcopy(lib)
        for cell in lib_copy.cells.values():
            cell.remove_polygons(lambda pts, layer, datatype: layer != keep)
            # Checking only the first item in path.layers works because the library was
            # loaded from a file, so it neve creates paths with more than 1 element.
            cell.remove_paths(lambda path: path.layers[0] != keep)
            cell.remove_labels(lambda label: label.layer != keep)
        for cell in lib_copy.top_level():
            # Make sure all cell names are valid file names
            cell.write_svg(f"{cell.name}-{keep}.svg")

    print("\n")
```

The bounding boxes will be different for each layer saved, since many elements are removed for each layer. If that's a problem, another option is to keep all elements, but make them transparent in the SVG output:

```python
import sys
import copy
import gdspy

if len(sys.argv) < 2:
    sys.stderr.write(f"Missing file name!\nUSAGE: {sys.argv[0]} filename\n")
    sys.exit(1)

for fname in sys.argv[1:]:
    print(f"Processing {fname} cells:", end="", flush=True)

    lib = gdspy.GdsLibrary(infile=sys.argv[1])
    for cell in lib.top_level():
        print(f" {cell.name}", end="", flush=True)

        ld, lt = cell.get_svg_classes()
        all_layers = set(layer for layer, typ in ld.union(lt))
        for keep in all_layers:
            style = {
                (layer, typ): {"stroke-opacity": "0", "fill-opacity": "0"}
                for layer, typ in ld
                if layer != keep
            }
            fontstyle = {
                (layer, typ): {"stroke-opacity": "0", "fill-opacity": "0"}
                for layer, typ in lt
                if layer != keep
            }
            cell.write_svg(f"{cell.name}-{keep}.svg", style=style, fontstyle=style)

    print("\n")
```

mo-nu, 2021-01-26:
This works great! I am getting files like these.
[AND2_X1.zip](https://github.com/heitzmann/gdspy/files/5875768/AND2_X1.zip)

Is there a way to remove the text/label which appears at the bottom of the images. 

Also, is it possible to get the layer names corresponding to the layer numbers.

heitzmann, 2021-01-26:
@mo-nu The layer names come from a typo in my script: it was `fontstye=style`, instead of `fontstyle=fontstyle`. It's fixed now.

Regarding layer names: GDS files have no notion of layer names. That information is not present in the file, it is a template or technology specification used by your software. You'd have to manually assign names to layer number (based on a dictionary, for example):

```python
layer_names = {
    0: "layer 0",
    1: "layer 1",
    # etc.
}
```

Then, when saving the file inside the loop, use, for example:
```python
layer_name = layer_names[keep]  # Make sure all you layers have a name, or this will raise an error.
cell.write_svg(f"{cell.name}-{layer_name}.svg", style=style, fontstyle=fontstyle)
```

mo-nu, 2021-01-26:
Thank you, that change fixed the layers part.

import gdspy
import numpy
import sys

filename=''

del sys.argv[0]

try:
	filename=str(sys.argv[0])
except:
	print(f"Must provide a filename")
	exit()

for file in sys.argv:
	filename=str(file)
	print(f"Generating .svg files for {filename}")
	lib = gdspy.GdsLibrary(infile=filename)

	cellList = lib.cells
	for val in cellList:
		cell = cellList[val]
		print(f"Processing {cell}")
		cell.write_svg(f"{val}.svg")

If I run this code, I get the svg file of the cell(with all layers). I am attaching an example file.
[AND2_X1.zip](https://github.com/heitzmann/gdspy/files/5876640/AND2_X1.zip)

How do I change this code, so that I don't get the text/label which appears at the bottom of the image.

mo-nu, 2021-01-26:
Thank you, that change fixed the layers part.

> import gdspy
> import numpy
> import sys
>
> filename=''
>
> del sys.argv[0]
>
> try:
> filename=str(sys.argv[0])
> except:
> print(f"Must provide a filename")
> exit()
>
> for file in sys.argv:
> filename=str(file) [...]

If I run this code, I get the svg file of the cell(with all layers). I am attaching an example file.
AND2_X1.zip

How do I change this code, so that I don't get the text/label which appears at the bottom of the image.

>

mo-nu, 2021-01-26:
Thank you, that change fixed the layers part.

```
import gdspy
import numpy
import sys

filename=''

del sys.argv[0]

try:
	filename=str(sys.argv[0])
except:
	print(f"Must provide a filename")
	print(f"Usage example: gdstosvg ../path_to/gds_files/*/*latest.gds")
	exit()

for file in sys.argv:
	filename=str(file)
	print(f"Generating .svg files for {filename}")
	lib = gdspy.GdsLibrary(infile=filename)

	cellList = lib.cells
	for val in cellList:
		cell = cellList[val]
		print(f"Processing {cell}")
		cell.write_svg(f"{val}.svg")
```

If I run this code, I get the svg file of the cell(with all layers). I am attaching an example file.
AND2_X1.zip

How do I change this code, so that I don't get the text/label which appears at the bottom of the image.

heitzmann, 2021-01-27:
@mo-nu In this specific case, it looks like all undesired labels are in layer 63, text type 63, so removing those from the library before starting the conversion should do the trick:
```python
lib = ...
for cell in lib.cells.values():
    cell.remove_labels(lambda label: label.layer == 63 and label.texttype == 63)
...
```

mo-nu, 2021-02-01:
Thank you so much!

closed 2021-02-01 by heitzmann