heitzmann/gdspy #168: Artifacts in GDSPY boolean operations

andrewdunsworth, 2021-01-26:
Thank you for this module, it is great.

I have noticed I can get artifacts that occur when doing booleans that involve a `PolygonSet` with many vertices, like this:

[image]

Sometimes, increasing the `max_points` of the boolean operation fixes the artifact, but other times it does not... 

Have you seen similar artifacts?  Do you have a robust solution / work around?

andrewdunsworth, 2021-01-27:
I found the source of the problem on my end, the way we generated the geometries were leaving <1 nm gaps between the ends of some things and the beginnings of the others.  This flaky rounding error was leading to bad polygon generation...

heitzmann, 2021-01-27:
@andrewdunsworth Yes, that would definitely be captured in the boolean operation. Glad you found it.

closed 2021-01-27 by heitzmann