heitzmann/gdspy #169: Please advise how reduce level of details in large GDS to make SVG of reasonable size

WhitemanE0, 2021-02-02:
Dear Sir,
Sorry for opening this in the issue section, but I know no other way to contact you.
I really need your advice. 
I need to reduce the level of details in GDS, let say, of 6Gb containing lot of identical cells in the hierarchy. Since the SVG is flat it became hundreds of Gigabytes. I need SVG for presentational purposes, so I don't need lot of details in nanometer scale.
If you provide me some starting points for development, I will really appreciate it.
Also please take into consideration, that I advertise GDSPY to my colleagues across the globe in a very good manner, as I believe it is really worthful.

WhitemanE0, 2021-02-02:
I see the following:
- remove some bottom hierarchy levels
- remove some unsignificant layers
- merge polygons

But high impact can give to remove shapes or part of the shapes with the area less than certain margin.

heitzmann, 2021-02-02:
Hello @WhitemanE0!  One thing to note is that the generated SVG file is NOT flat. It uses (almost) the same hierarchy the the GDSII does, so I strongly suggest not flattening any cells before creating the SVG. That will save a lot of memory.

However, when you display the SVG, that's another issue, since you'll be drawing all shapes on the screen you'll probably want to remove small or insignificant features. Your options make sense: remove as many insignificant references and layers as possible. Merging polygons might give you some gain, but only if you have a lot of overlapping geometry. Lastly, removing polygons with area below a certain threshold (before merging, if that's the case) should give good results if you can delete a reasonable amount of shapes. That should be the case if you're trying to create a preview of a whole layout (small details will be invisible anyways). Otherwise, if you want to preview a specific region of interest, you might be able to remove any shapes that are outside this region (with the `inside` function) and crop the remaining (with the `slice` function).

It's hard to give more specific advice without knowing the exact application of the SVG.

closed 2021-06-10 by heitzmann