heitzmann/gdspy #170: How to install the gdspy package in version 1.4.1

lhjey, 2021-02-06:
Hi, I try to install gdspy package in version 1.4.1 with command : pip install gdspy==1.4.1, while there is something wrong. So I want to know how to install the version 1.4.1 correctly. 
Thanks!

heitzmann, 2021-02-06:
Install information is in the README file. You didn't mention what OS you're using and what is the error you are seeing, but if you're on Windows, use a precompiled binary. To use pip and compile it yourself you'll need to install many other dependencies. Another option is to use Anaconda; there's a gdspy package for Anaconda that should work on Windows.

lhjey, 2021-02-06:
[image]

Defaulting to user installation because normal site-packages is not writeable
ERROR: Could not find a version that satisfies the requirement gdspy==1.4.1
ERROR: No matching distribution found for gdspy==1.4.1

heitzmann, 2021-02-06:
Strange, for some reason, minor versions from 1.4 are [missing from pypi](https://pypi.org/project/gdspy/#history).

You can download directly from the [1.4.1 release page](https://github.com/heitzmann/gdspy/releases/tag/v1.4.1) or give the url to [pip](https://pip.pypa.io/en/stable/reference/pip_install/):
```sh
python -m pip install --user https://github.com/heitzmann/gdspy/archive/v1.4.1.tar.gz
```

lhjey, 2021-02-06:
![image](https://user-images.githubusercontent.com/49546927/107116844-0c4ff980-68b1-11eb-8d64-8cf5c6861aab.p
Is it installation successful?The version displayed is still 1.4.

lhjey, 2021-02-06:
[image]

lhjey, 2021-02-06:
[image]

Is it installation successful?The version displayed is still 1.4.

heitzmann, 2021-02-06:
Yes, this was a problem with the 1.4 release… The version number was stuck at "1.4", but you have the right version.
That's probably the reason why these versions are not available in pypi.

lhjey, 2021-02-06:
OK, thanks for your help!

closed 2021-02-06 by heitzmann

anujashok194, 2022-02-11:
[image]
[image]

Can you please tell me what is the problem?