heitzmann/gdspy #173: Are two-element tuple ends like ('flush', 'round') allowed in FlexPath?

danielflanigan, 2021-04-08:
I'd like to draw a FlexPath with different end cap types at the start and end. Docs:
> ends ('flush', 'extended', 'round', 'smooth', 2-tuple, callable, list) – Type of end caps for the paths. A 2-element tuple represents the start and end extensions to the paths.

```python
import gdspy

library = gdspy.GdsLibrary(name='test_gdspy_ends')
cell = library.new_cell(name='main')

path = gdspy.FlexPath(points=[(0, 0), (100, 100)], width=1, ends=('round', 'flush'))
cell.add(element=path)
gdspy.LayoutViewer(library)
```
Gives
`numpy.core._exceptions.UFuncTypeError: ufunc 'multiply' did not contain a loop with signature matching types (dtype('<U32'), dtype('<U32')) -> dtype('<U32')
`
because numpy is trying to multiply the strings. Is the above syntax supposed to be allowed?

heitzmann, 2021-04-08:
No, that's not supported. The “extensions” you quoted mean numerical values representing how much the path extends on each end. The easiest way to get what you want is to manually insert a half circle at the start of your path and leave it with `ends="flush"`. You could also create a custom function that caps each end differently and use that (`ends=myfunc`).

danielflanigan, 2021-04-12:
I see. Thanks for the quick response.

closed 2021-04-12 by danielflanigan