heitzmann/gdspy #181: write_svg number resolution

horror-vacui, 2021-06-08:
Not a bug, just a feedback. SVGs could be smaller, and maybe faster to open.

The generated SVG file uses three decimal places behind the decimal point, but some of them have many additional ones. It is due to the number storage in a binary system, but displaying in decimal. I believe the SVG size and the speed of viewing it could benefit from consequently writing out only a given number of decimal places.

An excerpt from the generated svg file:
```
<polygon class="l15d0" points="-8.395000000000001,5.0649999999999995 -8.395000000000001,5.025 -9.935,5.025 -9.935,5.0649999999999995"/>
<polygon class="l15d0" points="-10.475,4.1450000000000005 -10.435,4.1450000000000005 -10.435,4.065 -10.475,4.065"/>
<polygon class="l15d0" points="-10.174999999999999,4.1450000000000005 -10.135,4.1450000000000005 -10.135,4.025 -10.174999999999999,4.025"/>
<polygon class="l15d0" points="-10.135,4.0649999999999995 -10.135,4.025 -10.225,4.025 -10.225,4.0649999999999995"/>
<polygon class="l15d0" points="-7.375,4.025 -7.375,4.0649999999999995 -7.2700000000000005,4.0649999999999995 -7.2700000000000005,4.025"/>
<polygon class="l15d0" points="-6.7749999999999995,4.1450000000000005 -6.735,4.1450000000000005 -6.735,4.025 -6.7749999999999995,4.025"/>
<polygon class="l15d0" points="-6.795000000000001,4.0649999999999995 -6.795000000000001,4.025 -6.875,4.025 -6.875,4.0649999999999995"/>
<polygon class="l15d0" points="-7.074999999999999,4.1450000000000005 -7.035,4.1450000000000005 -7.035,4.045 -7.074999999999999,4.045"/>
```

heitzmann, 2021-06-08:
I don't see how to fix the precision without possibly causing other problems. If we fix to 3 (`.3f`), for example, then coordinates below 0.001 will be lost. Who's to say they are not important? If we use a large enough number of places for a reasonable resolution (I don't really know what that would be, maybe 6?), then *all* coordinates will use that number of digits, which might end up increasing the total file size.

Any ideas?

horror-vacui, 2021-06-08:
What if this would be a argument for the write_svg function? In that case the user could tune it. The default behaviour could stay as it is now. I think numpy's format_float_positional function does the trick. What do you think?

```
>>> import numpy as np
>>> np.format_float_positional(np.pi, trim='-', precision=3)
'3.142'
>>> np.format_float_positional(np.pi*100, trim='-', precision=3)
'314.159'
```

horror-vacui, 2021-06-09:
It seems good to me. Though I am no expert. Thanks!

closed 2021-06-09 by horror-vacui