heitzmann/gdspy #182: error when using gdspy.Path.smooth()

zihegao, 2021-06-09:
I encountered the following error when I used `gdspy.Path.smooth()`

```
 File "C:\Users\Anaconda3\envs\layout\lib\site-packages\gdspy\polygon.py", line 1992, in smooth
    cta, ctb = _hobby(points, angles, curl_start, curl_end, t_in, t_out, cycle)

NameError: name '_hobby' is not defined
```
I think it is reproducible by running
```
import gdspy
import numpy

gdspy.current_library = gdspy.GdsLibrary() 

lib = gdspy.GdsLibrary()
cell = lib.new_cell('FIRST')
path = gdspy.Path(0.5, (0, 0))
path.smooth([(0, 5), (5, 5), (5, 10)])
cell.add(path)

gdspy.LayoutViewer()
```
Am I doing something wrong? Is `gdspy.Path.smooth()` still supported?

heitzmann, 2021-06-09:
No, you're right! That escaped our tests! Should be fixed now.

closed 2021-06-09 by heitzmann

zihegao, 2021-06-09:
Thanks for the swift fix!!