heitzmann/gdspy #183: unexpected behavior with negative coordinate values in gdspy.Path.smooth()

zihegao, 2021-06-09:
It seems when I use negative coordinate values for the points in `gdspy.Path.smooth()`, the result is unexpected. I was expecting path1 and path2 to look the same, except for the (-10,-10) offset.

```
import gdspy
import numpy

gdspy.current_library = gdspy.GdsLibrary() 

lib = gdspy.GdsLibrary()
cell = lib.new_cell('FIRST')
path1 = gdspy.Path(0.5, (0, 0))
path1.smooth([(0, 5), (5, 5), (5, 10)])
cell.add(path1)

path2 = gdspy.Path(0.5, (0-10, 0-10))
path2.smooth([(0-10, 5-10), (5-10, 5-10), (5-10, 10-10)])
cell.add(path2)

gdspy.LayoutViewer()
```
[image]

heitzmann, 2021-06-09:
@zihegao By default, [`Path.smooth`](https://gdspy.readthedocs.io/en/stable/reference.html#gdspy.Path.smooth) uses relative coordinates. You can fix your issue by using:
```python
path2.smooth([(0-10, 5-10), (5-10, 5-10), (5-10, 10-10)], relative=False)
```
Or, even better: by using relative coordinates you can use the exact same list of points for both paths:
```python
import gdspy
import numpy

gdspy.current_library = gdspy.GdsLibrary() 

lib = gdspy.GdsLibrary()
cell = lib.new_cell('FIRST')

points = [(0, 5), (5, 5), (5, 10)]

path1 = gdspy.Path(0.5, (0, 0))
path1.smooth(points)
cell.add(path1)

path2 = gdspy.Path(0.5, (-10, -10))
path2.smooth(points)
cell.add(path2)

gdspy.LayoutViewer()
```

zihegao, 2021-06-10:
@heitzmann thank you for the swift reply! Somehow I didn't realize that `relative=True` was default.

closed 2021-06-10 by zihegao