heitzmann/gdspy #184: gdspy installation problem

gxyuta, 2021-06-11:
Hello developer, I just tried to install the gdspy in py27 environment. However, I can't install the package due to lack of visual c++ 9.0 compiler. It is ok to install in py38 environment. I searched whole day for any way to install the 9.0 compiler. However, I didn't find any useful way. Do you have any experience for this problem.  Here is my error information:

```
(py27env) C:\WINDOWS\system32>pip install gdspy
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Collecting gdspy
  Using cached https://files.pythonhosted.org/packages/c9/b6/75106a390f5b755d4f8b9cacbbd008c27443be2635bc988a6728ebca2921/gdspy-1.6.6.zip
Requirement already satisfied: numpy in c:\programdata\anaconda3\envs\py27env\lib\site-packages (from gdspy) (1.16.6)
Requirement already satisfied: future in c:\programdata\anaconda3\envs\py27env\lib\site-packages (from gdspy) (0.18.2)
Building wheels for collected packages: gdspy
  Building wheel for gdspy (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: 'C:\ProgramData\Anaconda3\envs\py27env\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'c:\\users\\87490\\appdata\\local\\temp\\pip-install-e03oty\\gdspy\\setup.py'"'"'; __file__='"'"'c:\\users\\87490\\appdata\\local\\temp\\pip-install-e03oty\\gdspy\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d 'c:\users\87490\appdata\local\temp\pip-wheel-cihwp4' --python-tag cp27
       cwd: c:\users\87490\appdata\local\temp\pip-install-e03oty\gdspy\
  Complete output (33 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-2.7
  creating build\lib.win-amd64-2.7\gdspy
  copying gdspy\curve.py -> build\lib.win-amd64-2.7\gdspy
  copying gdspy\gdsiiformat.py -> build\lib.win-amd64-2.7\gdspy
  copying gdspy\hobby.py -> build\lib.win-amd64-2.7\gdspy
  copying gdspy\label.py -> build\lib.win-amd64-2.7\gdspy
  copying gdspy\library.py -> build\lib.win-amd64-2.7\gdspy
  copying gdspy\operation.py -> build\lib.win-amd64-2.7\gdspy
  copying gdspy\path.py -> build\lib.win-amd64-2.7\gdspy
  copying gdspy\polygon.py -> build\lib.win-amd64-2.7\gdspy
  copying gdspy\viewer.py -> build\lib.win-amd64-2.7\gdspy
  copying gdspy\__init__.py -> build\lib.win-amd64-2.7\gdspy
  creating build\lib.win-amd64-2.7\gdspy\data
  copying gdspy\data\00.xbm -> build\lib.win-amd64-2.7\gdspy\data
  copying gdspy\data\01.xbm -> build\lib.win-amd64-2.7\gdspy\data
  copying gdspy\data\02.xbm -> build\lib.win-amd64-2.7\gdspy\data
  copying gdspy\data\03.xbm -> build\lib.win-amd64-2.7\gdspy\data
  copying gdspy\data\04.xbm -> build\lib.win-amd64-2.7\gdspy\data
  copying gdspy\data\05.xbm -> build\lib.win-amd64-2.7\gdspy\data
  copying gdspy\data\06.xbm -> build\lib.win-amd64-2.7\gdspy\data
  copying gdspy\data\07.xbm -> build\lib.win-amd64-2.7\gdspy\data
  copying gdspy\data\08.xbm -> build\lib.win-amd64-2.7\gdspy\data
  copying gdspy\data\09.xbm -> build\lib.win-amd64-2.7\gdspy\data
  copying gdspy\data\down.xbm -> build\lib.win-amd64-2.7\gdspy\data
  copying gdspy\data\outline.xbm -> build\lib.win-amd64-2.7\gdspy\data
  copying gdspy\data\up.xbm -> build\lib.win-amd64-2.7\gdspy\data
  running build_ext
  building 'gdspy.clipper' extension
  error: Microsoft Visual C++ 9.0 is required. Get it from http://aka.ms/vcpython27
  ----------------------------------------
  ERROR: Failed building wheel for gdspy
  Running setup.py clean for gdspy
Failed to build gdspy
Installing collected packages: gdspy
    Running setup.py install for gdspy ... error
    ERROR: Command errored out with exit status 1:
     command: 'C:\ProgramData\Anaconda3\envs\py27env\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'c:\\users\\87490\\appdata\\local\\temp\\pip-install-e03oty\\gdspy\\setup.py'"'"'; __file__='"'"'c:\\users\\87490\\appdata\\local\\temp\\pip-install-e03oty\\gdspy\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'c:\users\87490\appdata\local\temp\pip-record-b5yhqj\install-record.txt' --single-version-externally-managed --compile
         cwd: c:\users\87490\appdata\local\temp\pip-install-e03oty\gdspy\
    Complete output (33 lines):
    running install
    running build
    running build_py
    creating build
    creating build\lib.win-amd64-2.7
    creating build\lib.win-amd64-2.7\gdspy
    copying gdspy\curve.py -> build\lib.win-amd64-2.7\gdspy
    copying gdspy\gdsiiformat.py -> build\lib.win-amd64-2.7\gdspy
    copying gdspy\hobby.py -> build\lib.win-amd64-2.7\gdspy
    copying gdspy\label.py -> build\lib.win-amd64-2.7\gdspy
    copying gdspy\library.py -> build\lib.win-amd64-2.7\gdspy
    copying gdspy\operation.py -> build\lib.win-amd64-2.7\gdspy
    copying gdspy\path.py -> build\lib.win-amd64-2.7\gdspy
    copying gdspy\polygon.py -> build\lib.win-amd64-2.7\gdspy
    copying gdspy\viewer.py -> build\lib.win-amd64-2.7\gdspy
    copying gdspy\__init__.py -> build\lib.win-amd64-2.7\gdspy
    creating build\lib.win-amd64-2.7\gdspy\data
    copying gdspy\data\00.xbm -> build\lib.win-amd64-2.7\gdspy\data
    copying gdspy\data\01.xbm -> build\lib.win-amd64-2.7\gdspy\data
    copying gdspy\data\02.xbm -> build\lib.win-amd64-2.7\gdspy\data
    copying gdspy\data\03.xbm -> build\lib.win-amd64-2.7\gdspy\data
    copying gdspy\data\04.xbm -> build\lib.win-amd64-2.7\gdspy\data
    copying gdspy\data\05.xbm -> build\lib.win-amd64-2.7\gdspy\data
    copying gdspy\data\06.xbm -> build\lib.win-amd64-2.7\gdspy\data
    copying gdspy\data\07.xbm -> build\lib.win-amd64-2.7\gdspy\data
    copying gdspy\data\08.xbm -> build\lib.win-amd64-2.7\gdspy\data
    copying gdspy\data\09.xbm -> build\lib.win-amd64-2.7\gdspy\data
    copying gdspy\data\down.xbm -> build\lib.win-amd64-2.7\gdspy\data
    copying gdspy\data\outline.xbm -> build\lib.win-amd64-2.7\gdspy\data
    copying gdspy\data\up.xbm -> build\lib.win-amd64-2.7\gdspy\data
    running build_ext
    building 'gdspy.clipper' extension
    error: Microsoft Visual C++ 9.0 is required. Get it from http://aka.ms/vcpython27
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'C:\ProgramData\Anaconda3\envs\py27env\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'c:\\users\\87490\\appdata\\local\\temp\\pip-install-e03oty\\gdspy\\setup.py'"'"'; __file__='"'"'c:\\users\\87490\\appdata\\local\\temp\\pip-install-e03oty\\gdspy\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'c:\users\87490\appdata\local\temp\pip-record-b5yhqj\install-record.txt' --single-version-externally-managed --compile Check the logs for full command output.
```

heitzmann, 2021-06-11:
Sorry, I only use linux. To me, Windows installation is a mess. I recommend you use the pre-compiled binaries found in the [releases page](https://github.com/heitzmann/gdspy/releases). I'm guessing in your case you'll use the [64 bit version for python 2.7](https://github.com/heitzmann/gdspy/releases/download/v1.6.6/gdspy-1.6.6-cp27-cp27m-win_amd64.whl).

tvt173, 2021-06-11:
Hi @gxyuta ,

I recommend on Windows installing via conda (rather than pip). Conda should automatically resolve these dependencies for you, so there is no need to go through the Visual Studio compiler installation at all.

I also recommend using python 3 instead of python 2, if possible, as python 2 is officially deprecated.

tvt173, 2021-06-11:
here is the download page for miniconda: https://docs.conda.io/en/latest/miniconda.html

gxyuta, 2021-06-11:
> Sorry, I only use linux. To me, Windows installation is a mess. I recommend you use the pre-compiled binaries found in the [releases page](https://github.com/heitzmann/gdspy/releases). I'm guessing [...]

Thanks for the help, I installed gdspy through the .whl file. There is another strange thing that if I use jupyter notebook in the py27 environment, I can't import both pygds and pyCAD shown in my pip list.

```
ImportError                               Traceback (most recent call last)
<ipython-input-3-dba9300b22d4> in <module>()
----> 1 import gdspy
      2 import math
      3 from gdsCAD import *
      4 import numpy as np
      5 wg=core.Elements()

ImportError: No module named gdspy
```
```
(py27env) C:\Users\87490\OneDrive - University of Texas at Arlington\1-Research\1-E-beam\gdsCAD_learn_11102020\Spiral_waveguide>pip list
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Package      Version
------------ -------------------
certifi      2020.6.20
future       0.18.2
gdsCAD       0.4.5
gdspy        1.6.6
numpy        1.16.6
pip          19.3.1
setuptools   44.0.0.post20200106
wheel        0.36.2
wincertstore 0.2

(py27env) C:\Users\87490\OneDrive - University of Texas at Arlington\1-Research\1-E-beam\gdsCAD_learn_11102020\Spiral_waveguide>
```

gxyuta, 2021-06-11:
> here is the download page for miniconda: https://docs.conda.io/en/latest/miniconda.html

Thanks for the help, I didn't know the difference between conda install and pip install. I tried to run `conda install gdspy`. Here is the error I got.

```
(py27env) C:\Users\87490\OneDrive - University of Texas at Arlington\1-Research\1-E-beam\gdsCAD_learn_11102020\Spiral_waveguide>conda install gdspy
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.

PackagesNotFoundError: The following packages are not available from current channels:

  - gdspy

Current channels:

  - https://repo.anaconda.com/pkgs/main/win-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/win-64
  - https://repo.anaconda.com/pkgs/r/noarch
  - https://repo.anaconda.com/pkgs/msys2/win-64
  - https://repo.anaconda.com/pkgs/msys2/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
```

tvt173, 2021-06-11:
Try this

conda install -c conda-forge gdspy

gxyuta, 2021-06-11:
> Try this conda install -c conda-forge gdspy
> […](#)

Sorry, new error:

(base) C:\Users\87490\OneDrive - University of Texas at Arlington\1-Research\2-Technology\E-beam\gdsCAD_learn_11102020\Spiral_waveguide>conda activate py27env

```
(py27env) C:\Users\87490\OneDrive - University of Texas at Arlington\1-Research\2-Technology\E-beam\gdsCAD_learn_11102020\Spiral_waveguide>conda install -c conda-forge gdspy
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: -
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
Examining @/win-64::__win==0=0:  80%|█████████████████████████████████████████▌          | 4/5 [00:01<00:00,  3.17it/s]\failed

UnsatisfiableError: The following specifications were found
to be incompatible with the existing python installation in your environment:

Specifications:

  - gdspy -> python[version='>=3.6,<3.7.0a0|>=3.7,<3.8.0a0|>=3.8,<3.9.0a0|>=3.9,<3.10.0a0']

Your python: python=2.7

If python is on the left-most side of the chain, that's the version you've asked for.
When python appears to the right, that indicates that the thing on the left is somehow
not available for the python version you are constrained to. Note that conda will not
change your python version to a different minor version unless you explicitly specify
that.
```

tvt173, 2021-06-11:
Looks like only python 3 is supported by the conda package. You'll either
need to upgrade to >3.6 or get it to install via pip/ the binary on python
2.7

closed 2022-01-14 by heitzmann