heitzmann/gdspy #189: How to change all layers in a cell ?

dorianherle, 2021-07-20:
Hi !

I would like to change all layers in an imported cell (from layer 101 to layer 0) for all polygons in that cell.
Here is a minimal working example of how I am doing it:

```
import gdspy

def change_layer_of_entire_cell(cell, layer):
    # get subcells of cell
    subcells = cell.get_dependencies()
    # if there are no dependencies -> no cell references
    if  subcells == set():
        # change cell directly
        subcells = [cell]
    
    for subcell in subcells:
        # get all polyogns in cell and attribute them to a specific layer
        extr_poly = subcell.get_polygons()
        # assign to layer
        polygons = gdspy.PolygonSet(extr_poly, layer=layer)
        # get all previous layers
        previous_layers = subcell.get_layers()
        print("previous_layers: " , previous_layers)
        # delete all polygons in cell
        for previous_layer in previous_layers:
            subcell.remove_polygons(lambda pts, layer_, datatype: layer_ == previous_layer)
            
        # add back polygons (on the correct layer)
        subcell.add(polygons)
    
# To reset the libary every time you run the code
gdspy.current_library = gdspy.GdsLibrary()
# Create a new library and set GDS unit to um, precision to nm
lib = gdspy.GdsLibrary(unit=1e-06, precision=1e-09)

# create test cell -> No references 
cell = gdspy.Cell('TEST_SIMPLE')
cell.add(gdspy.Rectangle((-40, -40), (40, 40), layer=101))
# my function to change layer of all polygons in cell
change_layer_of_entire_cell(cell, 0)

# create test cell -> With references 
cell_a = gdspy.Cell('A')
cell_a.add(gdspy.Rectangle((-40, -40), (40, 40), layer=101))

cell_b = gdspy.Cell('B')
ref = gdspy.CellReference(cell_a, (0,0))
cell_b.add(ref)

# my function to change layer of all polygons in cell
change_layer_of_entire_cell(cell_b, 0)

# view
gdspy.LayoutViewer(lib)
```

I was wondering if there is an easier option to do that? Thanks a lot! :) 
Amazing package by the way!

heitzmann, 2021-07-20:
@dorianherle You don't need to copy all polygons (or paths). It is possible to change the layers in the original objects:
```python
def change_layer_of_entire_cell(cell, layer):
    # Get dependency cells recursively
    all_cells = cell.get_dependencies(True)
    # Include original cell in the set
    all_cells.add(cell)
    for c in all_cells:
        # Process all polygons
        for polygon in c.polygons:
            # Substitute layer list for a new one with the
            # the same length and the desired layer number
            polygon.layers = [layer] * len(polygon.layers)
        # Process all paths
        for path in c.paths:
            path.layers = [layer] * len(path.layers)
        # Process all labels
        for label in c.labels:
            label.layer = layer
```
In your solution, the passed cell will only be included in the change list if it has no dependencies, which is probably not what you want, so the if statement there is incorrect.
You probably also want to get dependencies recursively, not to miss references of references, and so on.

dorianherle, 2021-07-21:
Thank you so much for your extremely quick reply and helpful code! It works perfectly. 
I added the ability to also change the datatype:

    
```
def change_layer_of_entire_cell(cell, layer, datatype=None):
    # Get dependency cells recursively
    all_cells = cell.get_dependencies(True)
    # Include original cell in the set
    all_cells.add(cell)
    for c in all_cells:
        # Process all polygons
        for polygon in c.polygons:
            # Substitute layer list for a new one with the
            # the same length and the desired layer number
            polygon.layers = [layer] * len(polygon.layers)
            # Proecessing datatype
            if datatype != None:
                polygon.datatypes = [datatype] * len(polygon.layers)
        # Process all paths
        for path in c.paths:
            path.layers = [layer] * len(path.layers)
        # Process all labels
        for label in c.labels:
            label.layer = layer
```

Thanks again!

closed 2021-07-21 by dorianherle

etha-gh, 2022-02-04:
Hi !

I'm analyze gds2 file in my school, 
So, I try to get coordinates in one meta layer's polygons.

now maybe I got something wrong.
here is my code.

I hope you give me advice or sample code to get coordinates in one meta layer's polygons.

Best regards

##########
import gdspy
import tripy
import pandas as pd
from plotnine import ggplot, aes, geom_polygon, geom_segment, geom_point

lib = gdspy.GdsLibrary(infile='test.gds')

###########################################################3
def convert_all(d2_array):
    ret_array = []
    for i in range(0, len(d2_array)):
        for j in range(0, len(d2_array[i])):
            ret_array.append(tuple(d2_array[i][j]))

    return ret_array

def convert_devide(d2_array):
    ret_array = []
    for i in range(0, len(d2_array)):
        ret_array2 = []
        for j in range(0, len(d2_array[i])):
            ret_array2.append(tuple(d2_array[i][j]))

        ret_array.append(ret_array2)
    
    return ret_array

def make_frame_from_triangles(triangles):
    x_start = []
    x_end = []
    y_start = []
    y_end = []
    for triangle in triangles:
        for i, pt in enumerate(triangle):
            next_index = (i + 1) % 3
            x_start.append(pt[0])
            x_end.append(triangle[next_index][0])
            y_start.append(pt[1])
            y_end.append(triangle[next_index][1])
    df = pd.DataFrame({'x': x_start, 'y': y_start, 'xend': x_end, 'yend': y_end})
    return df
########################################################################################

cells_polygons = []
cells_polygons.append(convert_devide(lib.cells['FlexPath1'].get_polygons()))

triangles = []
for j in range(0, len(cells_polygons)):
    for i in range(0, len(cells_polygons[j])):
        triangles.append(tripy.earclip(cells_polygons[j][i]))    
    
all_tri = []
all_tri = convert_all(triangles)

df = make_frame_from_triangles(all_tri)
print((ggplot(df, aes(x='x', y='y')) + geom_segment(aes(x='x', y='y', xend='xend', yend='yend'))))

#####################################

Tiancaizhi, 2024-01-02:
Hello, I seem to have found a way to do this with a single line of code. You can have a try:
cell.flatten(single_layer=xxx, single_datatype=xxx)