heitzmann/gdspy #190: How to add a duplicate Cell and change the layer of the polygon in the Cell

ojipadeson, 2021-07-26:
I now have a source gds file with 1 cell and 1 polygon in it. I want to insert it twice into the target gds, the first time setting the layer of the polygon to 10 and the second time setting it to 20 without creating many different cells in the cell list of the target gds file. However, in my code, the copy method does not succeed in changing the layer of the polygon.

```
insert = source_top_cell.copy(source_top_cell.name, deep_copy=True)
adding_polygon = gdsii_top_cell.get_polygonsets()
for polygon_ in adding_polygon:
      polygon_.layers = [given_layer]
insert.add(adding_polygon)
target_top_cell.add(gdspy.CellReference(insert))
```

The above code is run on the second insertion.

Thanks for any feedback!

heitzmann, 2021-07-26:
@ojipadeson Do you want to add cell references of just 2 copies of the polygon to the target GDSII cell?

ojipadeson, 2021-07-26:
@heitzmann I has just got some additional information and I can now use the `flatten()` function to meet the demands. Sorry to disturb you and thank you very much for your reply!

closed 2021-07-26 by heitzmann